OpenEtherCATsociety/SOEM #617: Sync Manager 7 does not get configurated

lbckmnn, 2022-05-24:
Hello,
I tried to use a slave which uses Sync Manager 7 for inputs.
Unfortunately no FMMU seems to be assigned to this Sync Manager.

I tracked this down to the following lines in ecx_config_create_input_mappings:

```/* search for SM that contribute to the input mapping */
while ((SMc < (EC_MAXSM - 1)) && (FMMUdone < ((context->slavelist[slave].Ibits + 7) / 8)))
{
     ...
     /* do configuration */
     ...
}
 ```
What is the reasoning of the (EC_MAXSM - 1)?
When I change this to just EC_MAXSM a FMMU is correctly assigned to the Sync Manger.

nakarlsson, 2022-06-09:
@lbckmnn , please create a PR and we can fix it.

lbckmnn, 2022-06-09:
@nakarlsson Yes but unfortunately I don't have access to the slave which needs sync manager 7 any more. 
Therefore I cannot be completely sure that the PR works.

lbckmnn, 2022-06-18:
It worked as i just changed the first 
`while ((SMc < (EC_MAXSM - 1)) && (FMMUdone < ((context->slavelist[slave].Ibits + 7) / 8)))` 
to
`while ((SMc < EC_MAXSM ) && (FMMUdone < ((context->slavelist[slave].Ibits + 7) / 8)))` 
but there are some other places where `(EC_MAXSM - 1)` is used as condition which i don't quite fully understand yet.
Should i change these too?

nakarlsson, 2022-06-21:
Hi,
No, only those ones, one for inputs and one for outputs.
Please acivate "EC_PRINT" and run slaveinfo and post the console, it will help to verify the result of the change.

while ((SMc < EC_MAXSM) && (FMMUdone < ((context->slavelist[slave].Ibits + 7) / 8)))

while ((SMc < EC_MAXSM)) && (FMMUdone < ((context->slavelist[slave].Obits + 7) / 8)))

Thanks

PR #622 (2022-07-02): also configure the last SM
fixes #617

unfortunately i don't have access to a slave which actually uses SM7, so i was not able to test this specific commit/PR. 
However, i did exactly the same when i had access to a slave which used SM7 and it did work (see https://github.com/OpenEtherCATsociety/SOEM/issues/617)
commit 42ec9bf2 (2022-07-02): also configure the last SM (#622)

fixes #617
closed 2022-07-02 by nakarlsson