OpenEtherCATsociety/SOEM #625: How to use SOEM with ctx?

lbckmnn, 2022-07-13:
I want to use SOEM with EC_VER2 and the context based methods (ecx_...) but I am not sure how to properly create a context.
The only example I found is here: [fsoe demo]( https://github.com/rtlabs-com/fsoe-soem-demo/blob/master/test/linux/fsoe_sample/fsoe_sample.c)
However, this seems to be rather complicated. 
Is there a simpler way than creating all members of the context and assigning the addresses?
I guess this is necessary because SOEM does not allocate any memory?

Thanks in advance

nakarlsson, 2022-07-13:
Here is a sample which setup 2 ctx/masters. Still the same concept though but no FSoE.

https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/intime/ec_master/ec_master.c

————

One idea would be to create context utils.

One SIZEOF_CTX macro that sum up the sizes used. The macro can than be used to do malloc or a byte array of a full contex.

#define ECAT_SIZEOF_CTX (sizeof(ec_slavet) * EC_MAXSLAVE + int /*slavecount*/ + ec_groupt * EC_MAXGROUP + …

And

One would need to setup the context pointers pointing into the memory allocated, but that could be a contex ”init” function?

ecx_contextt myctx;
void * myctx_data = malloc(1* ECAT_SIZEOF_CTX);

ecx_init_ctx(&myctx, myctx_data);

ecx_init_ctx(ecx_contextt * ctx, void * data)
{
    uint8_t data_ptr = data;
    ctx->port = data_ptr;
    data_ptr += sizeof(* ctx->port);
    ctx-> slavelist = data_ptr;
    data_ptr += sizeof(* ctx->slavelist);
…
}

But have we gained anything, might still seem complex? 😅

lbckmnn, 2022-07-14:
Yes that would be possible and would at least provide some kind of abstraction. 
Stuff can be added to the context without breaking application code.

Another possible solution would maybe be something like a "context_holder" struct.
``` C
struct context_holder { //maybe there is a better name for the struct
  ecx_portt port; //Note: no pointer!
  ec_slave_t slavelist[EC_MAX_SLAVES];
  ...
  struct ecx_context ctx;
};
```
and a respective function for initialization.
That would require less messing around with memory and the context can still be easily created on the stack or dynamically on the heap.

I might try to do this if i have time in the next weeks.

nakarlsson, 2022-10-18:
@lbckmnn , can we close this issue?

lbckmnn, 2022-10-18:
yes, thanks.

closed 2022-10-18 by lbckmnn