OpenEtherCATsociety/SOEM #628: CSP motor control logic

Davidinos, 2022-07-20:
Goodmorning,
I'm succesfully using SOEM with an Ethercat Panasonic driver MINAS-A6. So far I've always worked with PWM controlled motors hence I'm not sure about the right approach for ethercat driver motor, for example for accomplishing a trapezoidal movement (acceleration, max velocity and deceleration). Here my PDOs list is reported:

TxPDOs:
   603Fh        Error code
   6041h        Statusword
   6061h        Modes of operation display
   6064h        Position actual value
   606Ch        Velocity actual value
   6077h        Torque actual value
   60B9h        Touch probe status
   60BAh        Touch probe pos1 pos value
   60FDh        Digital inputs

RxPDOs:
   6040h        Controlword
   6060h        Mode of operation
   6071h        Target Torque
   6072h        Max Torque
   607Ah        Target Position
   6080h        Max motor speed
   60B8h        Touch Probe function
   60FFh        Target Velocity
   60B0h        Position Offset

Consider that in CSP mode Target Velocity is not used. For moving the motor with a trapezoidal movement I thought two different approaches:
1. Keep the Max motor speed constant and modify only the target position. Clearly in the acceleration phase the increment is progressively increased, in the max velocity phase the increment is kept steady and in the deceleration phase is decremented up to the final target position.
2. In the second approach, the target position is set equal to the final position from the very start and the several phases (acc, dec, max speed) are controlled using Max motor Speed only.

My questions are:
- What is the most suitable approach?
- can this approach be used for different movimentation types such as following another motor encoder as well?

I know that this topic doesn't concern soem but it'd be very useful for me. 
Thank you in advanced.
Regards,
Davidino

ArthurKetels, 2022-07-20:
In CSP mode the strategy is to deliver a constant stream of position values which the drive should track as closely as possible given the constraints of maximum velocity and maximum torque. The assumption is that the time interval between two successive position target values is so small that the movement is continuous.

This mode works and is used in many solutions. However the downside is that you have very little control over velocity and torque. Because when you limit those values too much the position controller is no longer able to track position. When you set the limits too high the movement becomes less smooth (resembles a stepper motor). If this behaviour is acceptable then you have found your solution.

A mode with full control is CST. The application generates a trajectory (for one or multiple axes) and computes a suitable torque for each motor. A feedback loop stabilizes the velocity and position. It is more work, and needs a better understanding of the physics involved, but has much better performance.

nakarlsson, 2022-10-18:
@Davidinos  , can we close this issue?

Davidinos, 2022-10-18:
Yes, @nakarlsson, I can consider it solved. Thank you.

closed 2022-10-18 by Davidinos