OpenEtherCATsociety/SOEM #629: How to read/write 1-bit size variables via SDO

d-loret, 2022-07-21:
We currently use SOEM's `ecx_SDOread()`/`ecx_SDOwrite()` to read and write SDOs. The functions work with byte buffers. However, we recently started working with a sensor which has some 1-bit size variables in its object dictionary (Micro-Epsilon ILD1900 laser sensors). Does SOEM provide an SDO mechanism to interact with variables of less than a byte in size?

Examples of these variables are 0x3450:4 or 0x3800:18 in the ESI file that can be downloaded from [here](https://www.micro-epsilon.com/download/software/ESI-Device-Description-Datei-ILD1900-EtherCAT.zip).

I would appreciate any help or leads on how to handle this type of variables.

nakarlsson, 2022-07-21:
And read/write a single byte doesn’t work?

ArthurKetels, 2022-07-21:
According to the standard ETG1000.5 objects of Boolean, BIT1 .. BIT7 are encoded in one octet when accessed as a single subindex. For example BIT3 is encoded binary as 0bxxxxxnnn, where x = undefined. When multiple subindexes are accessed with CompleteAccess then bit objects with lengths < 8 are concatenated and cross octet boundaries. When an object of 8 bits or a multiple are encountered they are padded to the next octet boundary.

SOEM has full support of SDO read and write including the above mentioned encodings. When using ecx_SDOread/write the user has to do the bit packing/unpacking.

d-loret, 2022-07-22:
Thank you for your answers.

I did try reading/writing with a single byte. It seemed to work for some objects (i.e. working counter was not zero), but not all. A small correction in my original post: the variables are Boolean. The way I am trying to access them is through index access with an unsigned 8-bit integer in which I assume 1->True and 0->False.

Today I retrieved the SDO abort codes and they were all 0x08000020 ("Data can not be transmitted or saved in application" from sensor's documentation). Based on your explanation above, it seems that I am already accessing the objects correctly, so it might be an issue with the sensor.

@ArthurKetels, to make sure I understood your explanation of Complete Access with objects of different sizes, is the following example of how data would be arranged in a CA buffer correct?

| Subindex | Type |  Position in byte buffer |
| ---------- | ----- | -------------------------- |
| 0 | uint8 | Byte 0: all bits |
| 1 | BIT5 | Byte 1: bits 0-4 |
| 2 | BIT6 | Byte 1: bits 5-7 and Byte 2:bits 0-2, padding for Byte 2:bits 3-7 |
| 3 | uint16 | Byte 3:all bits and Byte 4:all bits |
| 4 | Boolean | Byte 5:bit 0|
| 5 | BIT1 | Byte 5: bit 1 |

And one last related question, for [ec_errort](https://openethercatsociety.github.io/doc/soem/structec__errort.html), if its field `Etype` is equal to `EC_ERR_TYPE_SDO_ERROR`, is the union member `AbortCode` guaranteed to be used over the structure member (which includes `ErrorCode`, `ErrorReg`, etc.)?

d-loret, 2022-07-22:
Thanks again for clarifying my questions.

> In ETG1000.5 a boolean encoded as a single octet has the values 0x00 or 0xff. A BIT1 encoded as single octet has the values 0x00 or 0x01. And when a boolean is encoded in CA packed format it behaves a BIT1. Do you get the logic of that?

I think I do. In that case, there is a discrepancy between the sensor's documentation and its ESI file, because the ESI defines the object as Boolean, but the documentation defines it as "BIT" with 1 being True, and 0 being False. The sensor does not support Complete Access.

nakarlsson, 2022-10-18:
@d-loret , can we close this issue?