OpenEtherCATsociety/SOEM #630: EtherCAT Master with WSL 2.0

aschaepper, 2022-07-22:
Hello together

We are trying to run the stack inside WSL 2 on windows 10, we try to communicate to a Beckhoff EtherCAT Coupler.
Does anyone of you have expierence with running it in WSL 2.0?

Greetings

Alex

ArthurKetels, 2022-07-22:
I can not say it is impossible, but I did not find any way to pass RAW ethernet packets over the WSL virtual interface. The other parts of SOEM work just fine. Just do not expect any performance improvements over the native windows executable.

nakarlsson, 2022-11-17:
Have you succeeded in getting it to work?
@aschaepper , can this issue be solved?

aschaepper, 2022-11-17:
Hello @nakarlsson 
Unfortunately not, whe then stopped to investigate. To be hones it was a lack of knowledge / understanding on our side how WSL 2.0 and or the linux internals work. Maybe someone with more skills could get it to work, we just used a dedicated Box resp. VM.

I will close this now thanks anyway!

closed 2022-11-17 by aschaepper