OpenEtherCATsociety/SOEM #632: FMMU read and write type error

ChenHades666, 2022-07-28:
Background:
I am using SOEM to control LEADSHINE's EtherCAT IO expansion board EM64DX-E1.

problem found:
In the FMMU setting of SOEM, the setting of Type is opposite to that of TwinCAT.

Wireshark capture file:
[EM64DX_E1_Wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9205924/EM64DX_E1_Wireshark.zip)

| FMMU  | FMMU register address | Log Start  | Log Length | Log StartBit | Log EndBit | Phys Start | Phys StartBit | Type     | Activate |
| ----- | -------------- | ---------- | ---------- | ------------ | ---------- | ---------- | ------------- | -------- | -------- |
| FMMU0 | 0x600(TwinCAT) | 0x01000000 | 0x0004     | 0x00         | 0x07       | 0x1800     | 0x00          | 0x02(WR) | 0x01     |
| FMMU0 | 0x600(SOEM)    | 0x00000000 | 0x0006     | 0x00         | 0x07       | 0x1C00     | 0x00          | 0x02(WR) | 0x01     |
| FMMU1 | 0x610(TwinCAT) | 0x01000000 | 0x0006     | 0x00         | 0x07       | 0x1C00     | 0x00          | 0x01(RD) | 0x01     |
| FMMU1 | 0x610(SOEM)    | 0x00000006 | 0x0004     | 0x00         | 0x07       | 0x1800     | 0x00          | 0x01(RD) | 0x01     |

Since I'm running SOEM on ZYNQ, to avoid problems with ZYNQ itself, I run the unmodified simple_test on the Windows platform and get the following prints:

[image]

What I print on ZYNQ (I have EC_DEBUG turned on):

[image]

Also attach the ESI file of EM64DX-E1:

[EM64DX-E1_infineon.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9206116/EM64DX-E1_infineon.zip)

ChenHades666, 2022-07-28:
When running TwinCAT, I can enter OP mode and control IO, but when running SOEM, I cannot enter OP mode, and EM64DX-E1 reports Watchdog_PD Error.

ArthurKetels, 2022-07-28:
This is a long standing bug in the slave firmware. Complete access is implemented wrong for object 0x1c00. This results in swapping the in and outputs for the FMMU. SOEM just follows the information given by the slave. TwinCAT works because it reads the info from the ESI file and not directly from the slave.

A simple work-around is to disable CA for the slave.
After ec_config_init() but before ec_config_map() you can do
ec_slave[slc].CoEdetails &= ~ECT_COEDET_SDOCA;
to clear the Complete Access bit in the CoE capabilities list.

ChenHades666, 2022-07-29:
Thanks for your reply!

According to what you said, I modified the simpletest function in SOEM\test\linux\simple_test.c as follows:

```c
void simpletest(char *ifname)
{
    int i, j, oloop, iloop, chk;
    needlf = FALSE;
    inOP = FALSE;

   printf("Starting simple test\n");

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);

         for (int cnt = 1; cnt <= ec_slavecount; cnt++)
         {
            if (ec_slave[cnt].eep_man == 0x00004321 && ec_slave[cnt].eep_id == 0x11100093)
            {
               printf("Found %s in slave[%d].\n", ec_slave[cnt].name, cnt);
               ec_slave[cnt].CoEdetails &= ~ECT_COEDET_SDOCA;
            }
         }

         ec_config_map(&IOmap);

         ec_configdc();

         printf("Slaves mapped, state to SAFE_OP.\n");
           
     ... ...
```

Then I get the following printed:

[image]

And Wireshark's packet capture:

[image]

[image]

This doesn't look like any improvement.

ChenHades666, 2022-07-29:
I suspect it's Leadshine's problem, because I can see a lot of SDO errors in both TwinCAT and SOEM.

[image]

[image]

ChenHades666, 2022-08-30:
After adding related entries in the configuration table, it can be successfully run.

closed 2022-08-30 by ChenHades666