OpenEtherCATsociety/SOEM #634: Emergency mailbox error code

monk200, 2022-08-12:
When running slaveinfo on a port I've seen the occasional message along the lines of `EMERGENCY slave:2 error:8130`. It looks like this is getting printed out by [line 313 of ethercatprint.c](https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercatprint.c#L313) and 8130 is the ErrorCode from [line 499 of ethercattype.h](https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercattype.h#L499). Is there any documentation for what this specific code actually means? I've also seen an ErrorCode of 5441 a few times.

nakarlsson, 2022-08-12:
You should be able to google that or consult the slaves manual. 
EMERGENCY messages is CanOpen and not realy used in EtherCAT, more of a legacy thing.

The ErrorCodes you provided was present is this file, not sure if it is relevant.
https://www.motorpowerco.com/media/filer_public/5b/cc/5bcccb79-3309-4026-aaa1-d5993ca6c553/man-can301ig.pdf

nakarlsson, 2022-10-14:
@monk200 , can we close this issue?

closed 2022-10-16 by monk200