OpenEtherCATsociety/SOEM #635: Confirmation about SYNC0, x1c32

joserubenh, 2022-08-20:
Hello, I would like to confirm the following.

A. Is this diagram accurate?.
[image]

B. Given:

     uint32 cycletime;
     int32 shifttime;
     ec_dcsync0(1, TRUE, cycletime, shiftime)`;

SYNC0 event time is when  ` ec_DCtime % cycletime == shiftime `; ?

B. Q: The optimal arrival time of the PDO exchange via ec_send_processdata() is between 
     a).` x1c32:06 +  x1c32:09` ,.  (Shift time)
     b) ` x1C32:05 - 1C33:06` (Input Latch)

C: `x1C32:12` (SM Event missed) will output how many cycles were not updated by a PDO exchange.
D: An event will be missed, thus added to  `x1C32:12`  when the PDO exchange happens before InputLatch.

Thank you very much.

PS. I have read #585 (and some others). Thank you.

nakarlsson, 2022-08-20:
A. Connected to your question D , I believe you’re looking for a diagram that include SM2? SM2 would correspond to ec_DCtime. And as in your diagram, ec_send_processdata will send the frame seen before SYNC0.

B. Yes

C. Yes

D. No, that don’t make sense. You want your outputs in place when SYNC0 happens.

E.  Consult the standard

F. Yes, if SM event does not occur in a single cycle.

G. If the SM2 event has not happened  when the SYNC0 event happens

joserubenh, 2022-08-21:
**A** Is this diagram accurate?

_9.1.3 SyncManager/Sync0
This mode is recommended for most applications when the Sync0-event is used for synchronization.
The output process data mapping is triggered by the SM2 Event and the ECAT_Application to set the
output values and start the input latch is started on Sync0 (see Figure 30: SM/Sync0 Synchronization).
With monitoring the SM2-event (before the Sync0-event occurs), the application ensures that there are
new target values available for each local cycle. If the SM2-event is too late to complete the
CalcAndCopy before the Sync0-Event occurs, the “SmEventMissed_

[image]

Source: https://download.beckhoff.com/download/document/io/ethercat-development-products/an_et9300_v1i8.pdf ((PAGE 79)).

`//Regarding writing process data`

**B** `ec_send_processdata();` will perform the SM2 cycle.
**C** The SM2 Event(2) will latch input values and copy those values once Sync0_B cycle starts.
**D** SM2Event(2) should always happen before Sync0_B. 

`//Regarding reading process data`

**E**  `ec_receive_processdata();` will perform the SM3 cycle.
**F**  depending on configuration, output values will either be copied at Sync0_A, or right after SM2Event(2). 

_Thank you very much for your time._

nakarlsson, 2022-08-22:
Hi,

ec_receive_processdata() is not related to SM3, ec_receive_processdata() is merely the receive function tightly connected to ec_send_processdata(). By it self it will not accomplish anything.

Eg. 
ec_send_processdata() -> send the frame that generate the SM2 event, in that event SM2 is written with outputs from the frame , SM3 are read for inputs that get stored in the frame. This happens on-fly on already present data.
ec_receive_processdata() -> simply does receive of the frame sent by ec_send_processdata() .

Otherwise you seems to get correct info.

Diagram shows, I'd say, the most commonly used setup.

SyncManager/Sync0 will synchronize data and application
Sync0 alone will not synchronize with data, as written in the text it might still jitter since it doesn't synchronize with SM2.

joserubenh, 2022-09-06:
Just if anyone is following this thread.
Some drivers require to setup a special address to activate Sync0 as the latching time, regardless of ec_dSync0 instuction.
My particular driver is set up by calling     `ec_SDOwrite(1, 0x0300, 0x00, true, sizeof(activate), &activate, EC_TIMEOUTRXM);`
Most of the time, the object required to command Sync0 is specified on the xml file in OpMode/Assignactivate: `<AssignActivate>#x0300</AssignActivate>`

nakarlsson, 2022-10-18:
@joserubenh , can we close this issue?

joserubenh, 2022-10-18:
Yes please. thank you.

closed 2022-10-19 by ArthurKetels