OpenEtherCATsociety/SOEM #636: EtherCat communication failed off Suddenly

WillyTuring, 2022-08-20:
Hi Arthur,
As i am implementing the ZeroMQ along with EtherCat, i encountered this problem.
[image]

What do you think the problem might be? Thanks in advance for your help.

nakarlsson, 2022-08-24:
could you post the wireshark?

WillyTuring, 2022-08-24:
Hi, nakarlsson
I just managed to solve the problem, i found out what caused it so i tackled it, but i'm sure what went wrong with Soem slave under the hood.

But as i'm working on building a several processes to interact with Soem master, so i thought i will leave that problem aside for now.

Speaking of which, i know Soem master process needs two threads(one for PDO exchange with EtherCat slave, one for detecting the communication stability) while building the communication(through redtest( ) function from the orginal sample) with slave; after establishing the communication, in order to maintain the process we could either put the main() function in sleep or run a while loop in it, so in this case there are essentially 3 threads in this process.

My question is do we have to keep "3 threads" alive? Since in this case the main() thread is literally useless after establishing the communication. 
Originally i wanted to put a while loop in this main() thread and then use it to interface with other processes, so the RT ecatthread() only deals with the main() thread, i managed to get in sync with the main() thread and other processes; but since i couldn't control when does OS will switch from main() to ecatthread(), this is a little bit worrying.

And there's might be another way of considering the problem, what will happen if i just put a while(1) loop in the main() function and do nothing there, will OS just "ignore" that thread completely? If that's the case then i'd rather to use the ecatthread() as the interface to get the command, instead of using the main() thread as the interface to other processes.

nakarlsson, 2022-08-25:
I'd think of one reason, not saying it is a must or even needed.

Ex.
1. PDO thread
2. Slave diagnostics thread
3. Main thread, setup the system and could start/synchronize thread 1 & 2. This thread could be a state machine. When done, it is as you say useless. But as in the sample, you might wanna handle the takedown of the system as well. The thread could then "wake up" and do a controlled takedown. So, this thread can idle, sleep, while (1) on low prio when system have been started and put to work when the system is suppose to shutdown.

WillyTuring, 2022-08-25:
Hi nakarlsson,
Thanks for sharing the thoughts.
That' what i'm currently running on, there are exactly 3 threads as you pointed out in the Soem master process.
Like you said, i just run a blank while(1) loop in the main() thread just to keep the process alive.
Multi-process implementation is going well, i guess i will come back late r to affirm if the critical-timing demand of PDO thread is meet all the time or not.
I will close this up now and post another one later if anything new.
Thanks again and have a nice day!

closed 2022-08-25 by WillyTuring