OpenEtherCATsociety/SOEM #637: Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Happiness-yuan, 2022-08-25:
Dear sir,
i am a new programmer and now i need to know the soem well for my work;i am using SOEM from master. compile is OK; 
But when i use the code own test app(simple_test), the result is 'Slave = 1 State=0x12 StatusCode = 0x001e: Invalid input configuration', i didnt modify the offical code. 
```c
// simple_test
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on ens33 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 4 0 0 0
Request operational state for all slaves
Calculated workcounter 1
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x001e : Invalid input configuration

Request init state for all slaves
End simple test, close socket
End program
```
slaveinfo:
```c
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on ens33 succeeded.
Time:1661437716.947 SDO slave:1 index:1a1a.00 error:06020000 The object does not exist in the object directory
1 slaves found and configured.
Calculated workcounter 1
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:AX58100_16HBI
 Output size: 0bits
 Input size: 32bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000001 ID: 00058100 Rev: 00020111
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   2 F:00010064 Type:4
 SM3 A:1400 L:   2 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMU1 Ls:00000002 Ll:   2 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```
What is the reason for "invalid input configuration" could be?

Happiness-yuan, 2022-08-25:
ESI file:[58100-Device-16bitHBI.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9425740/58100-Device-16bitHBI.zip), I don't know how to solve this problem.

ArthurKetels, 2022-08-26:
It looks like an issue with the slave firmware and EEPROM data. Can you post a wireshark capture of the simple_test run?
Also the supplied ESI file looks very unconventional. I think they mixed up decimal and hexadecimal numbers as DefaultValue.

Happiness-yuan, 2022-08-26:
> It looks like an issue with the slave firmware and EEPROM data. Can you post a wireshark capture of the simple_test run? Also the supplied ESI file looks very unconventional. I think they mixed up decimal and hexadecimal numbers as DefaultValue.

Thank you sir.Here is the wireshark capture of the simple_test run.
[simple_test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9434614/simple_test.zip)

Yes,you are right, my slaver is a development board with AX58100. It's ESI file is unconventional.But, It seems to good when i used TwinCAT to control my slaver board.If you can give me advice, i will be very grateful!

ArthurKetels, 2022-08-26:
If found issues in the slave COE implementation.
Index 0x1c00 should be
```
sub 00 = 0x04
sub 01 = 0x01
sub 02 = 0x02
sub 03 = 0x03
sub 04 = 0x04
```
What is actually read from the slave is:
```
sub 00 = 0x04
sub 01 = 0x01
sub 02 = 0x02
sub 03 = 0x04
sub 04 = 0x04
```

And index 0x1c13 should be:
```
sub 00 = 0x02
sub 01 = 0x1a01
sub 02 = 0x1a02
```
What is actually read from the slave is:
```
sub 00 = 0x02
sub 01 = 0x1a01
sub 02 = 0x1a1a
```

Because the slave advertises CA as available option it is used by SOEM. The firmware has some buffer copy bug that causes the before last byte in the CA assembly to be the same as the last byte. Hence the wrong interpretation by SOEM and the fail in configuration.

Fix the slave firmware and all will be good.

Happiness-yuan, 2022-08-29:
> If found issues in the slave COE implementation. Index 0x1c00 should be
>
> ```
> sub 00 = 0x04
> sub 01 = 0x01
> sub 02 = 0x02
> sub 03 = 0x03
> sub 04 = 0x04
> ```
>
> What is actually read from the slave is:
>
> ``` [...]

I have fixed my slave firmware. Thank you very much!

closed 2022-08-29 by Happiness-yuan