OpenEtherCATsociety/SOEM #638: SOEM and SOES with EVB-LAN9252-SPI and Raspberry Pi

sc610889477, 2022-08-29:
Goodmorning to all,
I'm using a MacBook as Ethercat Master and an Ethercat Slave which uses a EVB-LAN9252-SPI and a Raspberry Pi 3B. 
The slave info with -map reports the following:
```
$./slaveinfo en24 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en24 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:LAN9252-C2000 Sample Application
 Output size: 8bits
 Input size: 8bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000004d8 ID: 0000000d Rev: 00000001
 SM0 A:1000 L:   1 F:00090044 Type:3
 SM1 A:1200 L:   1 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   1 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1A00 Byte 0
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x3101:0x01 0x08 OCTET_STR(8) Output
  SM1 TXPDO 0x1600 Byte 0
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x3001:0x01 0x08 OCTET_STR(8) Input
End slaveinfo, close socket
End program
```
I want to use SOES's [raspberry_lan9252demo](https://github.com/OpenEtherCATsociety/SOES/tree/master/applications/raspberry_lan9252demo) to control LEDs and buttons connected to Raspberry Pi's GPIO. The [ESCREG_ALEVENT_SM2](https://github.com/OpenEtherCATsociety/SOES/blob/fc72fbeeecc10d53c9c0655c88d43214c4ef0759/soes/ecat_slv.c#L221) parameter is used when judging whether to use the button's function([cb_set_outputs()](https://github.com/OpenEtherCATsociety/SOES/blob/8de160a85159b1a4667004a4ec927389c1bee582/applications/raspberry_lan9252demo/main.c#L33)).

**Below is my train of thought, please correct me if I'm wrong.**

Above my slaveinfo SM2 doesn't seem to start, so I should edit the ESI file to configure SM2 and SM3.
I used siitool to successfully convert the xml file into a bin file, and then used SOEM's [eepromtool.c](https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/linux/eepromtool/eepromtool.c) to successfully write the bin file into the slave.
Unfortunately, the SDK of EVB-LAN9252-SPI is [LAN9252_C2000_SDK_V1.0](http://ww1.microchip.com/downloads/en/DeviceDoc/50002604A.pdf), and I did not find this SDK. The XML files I found on the web did not match my EVB-LAN9252-SPI. After forcibly writing, it will become the following result.
using [LAN9252_PIC32_SDK_v1.1](https://www.microchip.com/en-us/software-library/lan9252-ethercat-sdk)/ESI Files/SPI-withGPIO-2PortMode.xml
```
$ .slaveinfo en24 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en24 succeeded.
1 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:Microchip PIC32 Slaves
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: e00004d8 ID: 00009252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   4 F:00010064 Type:3
 SM3 A:1400 L:   4 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 01 FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
End slaveinfo, close socket
End program
```
running slample_test
```
$ ./simple_test en24
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on en24 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Operational state reached for all slaves.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
...
```

ArthurKetels, 2022-08-29:
This is not a master but slave question. Better post this in the SOES section. SOEM seems to work OK.

sc610889477, 2022-08-30:
> 

Ok, thanks for your reply. I will go to SOES to ask question.

closed 2022-08-30 by sc610889477