OpenEtherCATsociety/SOEM #639: Slave's long delay to initial torque command

WillyTuring, 2022-09-04:
Hi Arther,
Hi all,

My problem is just like the title described, I attached the data captured from Wireshark, let's take a look together.
[Wireshark_capture.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9483775/Wireshark_capture.zip)

As we can see at the packet #64076, after motor got enabled, i sent a target torque command(hex 32), but it took a whole lot of time for slave to respond to this initial target torque, i just don't understand what is happening here?

Does anyone have some thoughts about this issue? Thanks in advance for everyone's help.

ArthurKetels, 2022-09-04:
Hi Willy.

First of all, this is not a SOEM question but a slave behavior. Second, you make it very hard for us to follow your reasoning. A raw wireshark dump is not the right place to precisely track the PDO data of a slave.

These kind of questions are generally best left between user and manufacturer of the slave. How can we know how the slave firmware is build up? What I do know is that the bulk of servo drives respond very quickly to changes in setpoint, often within 1 or 2 cycles.

One though is perhaps the drive has a brake lifting delay, and only starts outputting torque after a fixed delay. But this should only be noticeable after first start of the motor. And this should be documented in the slave manual.

WillyTuring, 2022-09-04:
Hi Arthur,

Thanks for the help as always. I'm fully aware that this has nothing to do with Soem, just wanted to post it up here to see if anyone who has more experience on motor/driver control got any thoughts to share or discuss.

I actually followed your advice last time on maybe consider trying another driver, and i actually did, though not a big brand, and the response curve is still quite similar, nothing compares to what you just mentioned that "the bulk of servos can respond to setpoint change in 1 or 2 cycles", that seems a little bit impossible to me considering the "electric time constant" itself is a couple of milliseconds depending on the power of the motor, so when people are running EtherCat cycle at 1k or even 4k loop, i'm genuinely curious how did they achieve that?

And then Arthur you mentioned about "A raw wireshark dump is not the right place to precisely track the PDO data of a slave.", i'm wondering what's the appropriate way to post the question like this with the necessary raw information?

ArthurKetels, 2022-09-04:
When asking any of us to look into data it is YOUR job to represent that data in the most easy way to understand. You didn't even mention at what byte locations we have to look for setpoint and torque response. Further some plots will help a lot in understanding. But make the plots readable with notations on axis and cursors where appropriate. If it takes more than 1-2 minutes to understand the issue people will not respond to your questions. As a last step add the raw file for the nitty gritty details.

I think I need to be careful with the word "response". In control theory this has a very specific definition. What I though you where referring to is the latency (the time it takes for a new setpoint to be visible in the output). This should be quick and that is my reference above, 1-2 cycles. Then there is the "plant" response, this is dominated by inductance of the motor. If the drive is slow here, yes then there is a problem. A plot with the time in x-axis and setpoint and measured torque in the y-axis should show this clearly. Your first slave had the extra complication that the measured torque was heavily filtered. Then we can substitute measured torque by measured acceleration (2nd derivative of position).

The motor electrical time constant is TauE = L / R. Where L = inductance and R = resistance. This is the time where the current reaches 63% of its steady state value after a step change in voltage.

For the mechanical time constant of a motor this is:
TauM = R * J / Ke * Kt. Where R = resistance [Ohm], J = Inertia [kg*m^2], Ke = Torque constant [Nm/A], Kt = Motor constant [V/rad/sec].

To get a decent working servo controller : Latency < TauE < TauM. But most important is TauE < TauM. 

And yes, the selection of the motor is very important. It is no use running EtherCAT cycles at 4kHz when you TauE = 10ms.

WillyTuring, 2022-09-08:
Hi Arthur,
Sorry for the late response to your helpful insight, i just got back from a bereavement leave.

As i was changing the mapped object inside 0x1600, i always got some kind of PDO configuration error, the way i did this as follows as i tested in the "slaveinfo" source code:
1.I set the value in subindex 0 of object 0x1600 to 0;
2.then i modify from subindex 1 to subindex 7 one by one;
3.at the end i set the value in subindex 0 of object 0x1600 to 7;
I can see the changes being successfully made and print out in the with "-map", but when i moved to my source file, it just didn't work.

I wondered what did i miss here?

ArthurKetels, 2022-09-08:
I an sorry to hear you had to take that leave, I hope you are doing OK.

Changing the slave configuration is best done in the  preop to safeop hook, as is explained in the documentation (tutorial.txt).
`ec_slave[slc].PO2SOconfig = EL7031setup;`

The configuration and mapping logic of SOEM will then automatically take care of the slave configuration.

WillyTuring, 2022-09-08:
Hi Arthur,
I did use the PO2SOconfig method like this:
[image]
Where did i do wrong?

ArthurKetels, 2022-09-08:
You should re-read the tutorial.txt file. You should make a function that does the SDOwrites. F.e.:
```
int MYsetup(uint16 slave)
{
   .. do all the ec_SDOwrite stuff ..
}
```
and link it to the hook:
`ec_slave[6].PO2SOconfig = MYsetup;`

WillyTuring, 2022-09-09:
Hi Arthur,
I adopted the format you suggested above, but still failed.
[image]
[image]

I checked the " .xml " file, 0x1600 is allowed for manual change.
[image]

What's going on here?

ArthurKetels, 2022-09-09:
A wireshark capture of your last attempt would be useful. And if possible also a capture of your test program that did succeed to modify the mapping.

Also when something does not work with re-mapping please debug it so that you print the return value of each ec_SDOread/write. Then at least you know if the operation was successful or not.

WillyTuring, 2022-09-09:
Hi Arthur,
I attached the wireshark file and a capture of the code, as well as a capture of the print out.
[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9535379/wireshark.zip)
[code.zip](https://github.com/OpenEtherCATsociety/SOEM/files/9535381/code.zip)
[image]

ArthurKetels, 2022-09-09:
I do not understand what you think is wrong with your code. Both from the screen shot and wireshark capture it seems to work just fine. All slaves are configured and mapped correctly and go to OP state.

What I do notice is that you set 0x1600:00 back to 7 while you only program 6 indexes. Shouldn't you set 0x1600:00 to 6? Now the seventh subindex is still at the old value and included in the mapping.

WillyTuring, 2022-09-10:
Hi Arthur,
Entry No.7 in 0x1600 doesn't matter, i won't use it anyway, but yes you are right, i could be more cautious to make it 6 so that it becomes inaccessible.

Yes the slave can go to Op mode and the PDO entry object has indeed being changed, but the slave wouldn't execute any of my control command, like enabling the motor, as you can see starting from the wireshark packet No.63362, i was trying to enable motor with series command, but it got stuck.

WillyTuring, 2022-09-15:
Hi Arthur,
Sorry for bothering again, i'm aware you are busying, just want to check out quickly to see if you have got any thoughts yet on the problem i posed earlier.

ArthurKetels, 2022-09-15:
As this is a slave specific issue, and even more a slave I have never touched myself it is difficult to give advice. The EtherCAT master side seems to work as intended. I do know that some slaves are very picky in how you configure the PDO. And for no clear reason they refuse some combinations. You have to work with the vendor to figure out what is the cause of the problem.

WillyTuring, 2022-09-16:
Hi Arthur,
Thanks for the suggestion, i'll see what i can do.

closed 2022-09-16 by WillyTuring