OpenEtherCATsociety/SOEM #640: When Master  is will  Check State's Slaves?

tuhai711, 2022-09-14:
Hello,
I have problem when make Master to control Salves.
I see in simple_test had function ecatcheck, master allaway check status state's slave.
I have a question, when master need check state's slave?

nakarlsson, 2022-09-14:
For that particular case it will "check" if the expected working counter differ from the actual working counter.

https://github.com/OpenEtherCATsociety/SOEM/blob/42ec9bf238ae617068a3adafd4c659573bdf29ec/test/linux/simple_test/simple_test.c#L153

tuhai711, 2022-09-14:
Hello.
Does the master need to constantly check Salve status?

nakarlsson, 2022-09-14:
Yes that is best practice if you implement your own master. But how you implement it can be done differently.

tuhai711, 2022-09-14:
Yes, thank you.

tuhai711, 2022-09-14:
Hello again,
How to send  output ec_send_processdata only one slave?

tuhai711, 2022-09-14:
Yes, Thank you

tuhai711, 2022-09-15:
Hello again.
If i want only received ec_receive_processdata from slave, How to implement? Because i see need use both  ec_send_processdata then ec_receive_processdata, we will receive data from slave.

nakarlsson, 2022-09-15:
As I answered in https://github.com/OpenEtherCATsociety/SOES/issues/132#issuecomment-1247656800.
All traffic is driven by the master-

tuhai711, 2022-09-15:
Yes. Thank you.

closed 2022-09-15 by tuhai711