OpenEtherCATsociety/SOEM #644: EL3064, firmware 20/hardware 17 cannot be brought to SAFE_OP

akstrifork, 2022-09-20:
We have used lib soem before with older EL3064 - but this new version from April 2022 will not get past PRE_OP
We used an adapted [simple_test.c](https://github.com/OpenEtherCATsociety/SOEM/tree/master/test/linux/simple_test) to isolate the problem

Does anyone know if changes in the lifecycle using libsoem? We cannot derive any changes from https://download.beckhoff.com/download/Document/io/ethercat-terminals/el30xxen.pdf and Beckhoff support has not obvious suggestions

ArthurKetels, 2022-09-20:
Looks like a configuration error. Can you please share a Wireshark log of the simple_test run?

akstrifork, 2022-09-20:
Here is the output from the unmodified version of [simple_test.c](https://github.com/OpenEtherCATsociety/SOEM/tree/master/test/linux/simple_test) and wireshark dump

https://drive.trifork.com/index.php/s/mDip78k6eodaRio

nakarlsson, 2022-09-20:
Does the setup work with TwinCAT? Eg are there any HW issues?

akstrifork, 2022-09-21:
Beckhoff just send me this:
_The behavior is known.
There was a change in the latest firmware, which is completely inside the EtherCAT specifications, but many 3rd party slaves also our EKs had some problems with.

Therefor we created again a new firmware to change this behavior back.

Attached you can find the new firmware._

nakarlsson, 2022-09-21:
Make sense, the wireshark looked wierd where the EK1100 reported error.

Can we close this issue?

akstrifork, 2022-09-21:
Apparently new EL3064 firmware is on the limit of specifications, even the EK1100 is not able to communicate with it