OpenEtherCATsociety/SOEM #646: SOEM + Windows - Winpcap is receiving frames that were transmitted

craigsacco, 2022-10-01:
Environment:
- Windows 11 22H2
- Visual Studio 2022 17.3.5
- SOEM master (commit 26fc5dd8e3c8981b4e8f83d736e691417bebacdb)
- Winpcap 4.1.3
- Realtek 2.5GbE NIC
- Beckhoff EK1100 /w EL1809 + EL2809 + EL3356 + EL4004

I was able to compile all of the SOEM tools in this environment without any issues, but when I run slaveinfo.exe, it is reporting that there are no slaves present:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{F7DA0DF0-67A1-4195-96CF-F21F09EA032D} succeeded.
No slaves found!
End slaveinfo, close socket
End program
```

I decided to inspect the traffic with Wireshark and found that the WKC value for the BRD command is being returned correctly:

[image]

When inspecting the packet received by ecx_inframe(...), it turned out to be the frame that was sent by the master to the slave. The WKC field for any sent frame is always zero. This would be interpreted as being unable to detect any slaves on the interface.

One differentiator between sent and received frames (from the perspective of the master) is that received frames would have the LA (locally-administered) bit set in the source MAC address.

I decided to make a subtle change to the ecx_inframe(...) function where I would check if that bit is set AS WELL AS the ethernet type field - changing lines 411-412 of nicdrv.c from this:

```
/* check if it is an EtherCAT frame */
if (ehp->etype == htons(ETH_P_ECAT))
```

to this:

```
/* check if it is an EtherCAT frame, and if the LA bit in the MAC address is set (for a response frame) */
if ((ehp->etype == htons(ETH_P_ECAT)) && (ehp->sa0 & htons(0x0200)))
```

This time, I was able to query all slaves on the interface:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{F7DA0DF0-67A1-4195-96CF-F21F09EA032D} succeeded.
Time:1664597208.127 SDO slave:5 index:1c13.01 error:06090011 Subindex does not exist
5 slaves found and configured.
Calculated workcounter 6

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0

Slave:2
 Name:EL1809
 Output size: 0bits
 Input size: 16bits
 State: 4
 Delay: 145[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 07113052 Rev: 00110000
 SM0 A:1000 L:   2 F:00010000 Type:4
 FMMU0 Ls:0000000a Ll:   2 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0

Slave:3
 Name:EL2809
 Output size: 16bits
 Input size: 0bits
 State: 4
 Delay: 285[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1003
 Man: 00000002 ID: 0af93052 Rev: 00110000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 SM1 A:0f01 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 140[mA]
 only LRD/LWR:0

Slave:4
 Name:EL3054
 Output size: 0bits
 Input size: 128bits
 State: 4
 Delay: 425[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1004
 Man: 00000002 ID: 0bee3052 Rev: 00130000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00000004 Type:3
 SM3 A:1180 L:  16 F:00010020 Type:4
 FMMU0 Ls:0000000c Ll:  16 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 07 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 130[mA]
 only LRD/LWR:0

Slave:5
 Name:EL4004
 Output size: 64bits
 Input size: 0bits
 State: 4
 Delay: 575[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1005
 Man: 00000002 ID: 0fa43052 Rev: 00120000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   8 F:00010024 Type:3
 SM3 A:1180 L:   0 F:00000000 Type:4
 FMMU0 Ls:00000002 Ll:   8 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 03 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 190[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

I am not too sure if I have fixed a defect in SOEM with this change (in fact I'm sceptical on whether this is a defect as it would have been encountered by others), or if there is some quirk with Winpcap while running in promiscuous mode.

If I could get some guidance on whether this is a pre-existing defect with SOEM, or if there is something that I am missing in my environment, that would be greatly appreciated.

craigsacco, 2022-10-01:
For giggles, I did a read-up on Npcap and realised that it is a drop-in replacement for Winpcap.

I replaced Winpcap with Nmap 7.93 (which also installed Npcap 1.71) and reversed the code change in my previous post - slaveinfo was able to query all the slaves.

Now that this is no longer a code defect, There should be a documentation update on [doc](https://github.com/OpenEtherCATsociety/SOEM/tree/master/doc)/soem.dox that requires Winpcap for Windows 7/8, and Npcap on Windows 10/11

ArthurKetels, 2022-10-02:
This packet echo is a known problem with windows. It appeared and then disappeared a few times in the past. You will find some patches for this in the commit history. For reasons of uniformity amongst implementations I decided to drop the patch. Npcap is a better solution for win10/11 anyway. It is even likely that new versions of win11 will altogether lose the ability to send EtherCAT packets.

nakarlsson, 2022-11-17:
@craigsacco , can this issue be closed?

closed 2022-11-24 by ArthurKetels