OpenEtherCATsociety/SOEM #647: Can I read the desired firmware name from the slave station through an API?

yanzhang920817, 2022-10-08:
When I upgrade the firmware to each slave station through FOE, I first issue the firmware name to each slave station. After the slave station matches and verifies the name successfully, I can issue the firmware content to the slave station. Can I first read the desired firmware name of the slave through an API, then read the corresponding file, and then send it to the slave. Can it be realized? Can you give me an example? I can't find it here. Thank you.

ArthurKetels, 2022-10-09:
Although FoE is a bidirectional protocol (read/write) and you could even read a directory on the slave this is seldomly implemented. The common use is writing a firmware update file to a slave with a specific name and password (default = 0). So what you want to do as proposed above is not possible.

yanzhang920817, 2022-10-10:
> Although FoE is a bidirectional protocol (read/write) and you could even read a directory on the slave this is seldomly implemented. The common use is writing a firmware update file to a slave with a specific name and password (default = 0). So what you want to do as proposed above is not possible.

Now, because we have chips from different manufacturers, but the chips have the same function, when we update firmware for different chips, we will use different firmware names, but sometimes we forget which chip model it is. Therefore, we want to obtain the name of the previously written firmware through foe to determine what firmware to use.

nakarlsson, 2022-10-10:
You can probably determine that in another way? There is a type of ESC register.

yanzhang920817, 2022-10-10:
> You can probably determine that in another way? There is a type of ESC register.

a type of ESC register? Can you describe the general process?

nakarlsson, 2022-10-10:
[image]

ArthurKetels, 2022-10-10:
I think the OP did not refer to the ESC chip but the microcontroller they use. If you are the manufacturer of the EtherCAT slave it is rather trivial to implement the read functionality in FoE. An other solution is to have the chip encoded in your version identifier in object 0x1018. And the simplest solution is to check the firmware file name in your slave. If the file name does not match the internal chip type simply refuse. The firmware update software can then simply cycle through all options. For a few tens of variations this is still very fast.

Actually this is not a EtherCAT master question but a slave implementation question.

nakarlsson, 2022-10-18:
@yanzhang920817, any further questions? Or can we close this issue?

nakarlsson, 2022-11-17:
Close due to inactivity

closed 2022-11-17 by nakarlsson