OpenEtherCATsociety/SOEM #648: how to clear the slave error?

yanzhang920817, 2022-10-11:
hello,
for example，in the follow code，the result of rval & 0x000f is the slave status, the result of  **rval  & 0x00f0**  is the slave error code，how to clear the error?

`     slstat.alstatus = 0;
      slstat.alstatuscode = 0;
      ecx_FPRD(context->port, configadr, ECT_REG_ALSTAT, sizeof(slstat), &slstat, EC_TIMEOUTRET);
      rval = etohs(slstat.alstatus);
      context->slavelist[slave].ALstatuscode = etohs(slstat.alstatuscode);
      state = rval & 0x000f; /* read slave status */
`

nakarlsson, 2022-10-18:
The master acknowledges the Error Indication flag of the slave by setting the Error Ind Ack flag (0x0120.4).

nakarlsson, 2022-11-17:
@yanzhang920817 , can this issue be closed?

closed 2022-12-03 by yanzhang920817