OpenEtherCATsociety/SOEM #649: Windows Build with NMake

CCSpellman, 2022-10-13:
Hi,

I am a newbie using cmake and have a problem. Apologies if the answer is obvious.

I have a problem building SOEM for Windows.
I have Visual Studio 2017. So I think I have nmake installed but I am not sure.
I have cmake version 3.25.0-rc1 installed.

The first problem I have is that when I execute the command line below...

cmake .. -G "MinGW Makefiles"

I get the error below...

 CMake Error at CMakeLists.txt:3 (project):
  Running

   'nmake' '-?'

  failed with:

   The system cannot find the file specified

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "D:/DevInfo/EtherCat info/SOEM-Master/SOEM-master/SOEM-master/build/CMakeFiles/CMakeOutput.log".

When I inspect CMakeOutput.log I see the comment below.

The system is: Windows - 10.0.19044 - AMD64

Thanks in advance for any help.

nakarlsson, 2022-10-13:
Did you run from a Visual Studio Developer Command Prompt?

CCSpellman, 2022-10-14:
Hi Thanks for the reply.

I thought I had but it was not working correctly. Fixed now.

Example code compiled correctly and now have examples working.

Thanks for your help!!

closed 2022-10-14 by nakarlsson

Bipin9179, 2024-03-14:
Hi CCSpellman,
Can you guide me to create my first SOEM code please