OpenEtherCATsociety/SOEM #653: How to prove synchronization between master and slave ?

casinia, 2022-10-19:
I used the examples in red_test.c. My question is what data do I need to record to prove that the master and slave are perfectly synchronized, and how do I compare these data?

casinia, 2022-10-19:
The PDO cycle  is 1ms, so I record the difference between the DC time received in two consecutive cycles and observe the error between this difference and 1ms. Can the synchronization between the master station and slave station be verified in this way?

ArthurKetels, 2022-10-19:
Yes this an accurate way to measure the relative distances between the master and slave reference clock. For synchronization I often refer to my earlier post : https://github.com/OpenEtherCATsociety/SOEM/issues/487#issuecomment-786245585

casinia, 2022-10-20:
Thank you so much Arthur. 
I grabbed the ec_DCtime of two consecutive cycles on master, and then recorded the difference between ec_DCtime (x+1) and ec_DCtime (x). Here is the data I recorded. And I think master and slave have successfully synchronized, right?

[logged.txt](https://github.com/OpenEtherCATsociety/SOEM/files/9825977/logged.txt)

Plus, in question # 485, you mentioned "Then you need to place the transmission of the master PDO cycle in the time slot of the slave (again by slave manufacturer data). For example when the sync0 is fired every 1000us the PDO transmission reaches the slave at 500us offset“，what functions do I need to implement it?