OpenEtherCATsociety/SOEM #654: Correct way to send commands with EtherCAT and SOEM

pauladam316, 2022-10-19:
Hello!

I have a rather basic question but I've spent some time trying to understand the best way to do it. I'm trying to implement an interface where I can send commands from a master to a custom developed slave running SOES. the commands would do things like enable LEDs,  trigger functions on the slaves microcontroller, etc. I made it work by adding fields to the slaves input PDO that the slave would parse, but this doesn't seem like the "correct" way to do it, as the PDO can be missed by the slaves microcontroller and thus misses the command. The correct way to do this seems to be using SDOs and mailbox communication vs buffered communication. Can anyone point me in the direction of example code that does something like this already? Or help clarify for me what the correct way to implement commands to a slave would be?

nakarlsson, 2022-10-21:
Check ET1020 9. Command Object

excerpt
"
9 Command Object
The command object should be used for actions or services (commands in the following) which cannot
be adapted to a single SDO Upload or SDO Download service; for example if the command needs
request and response data or if the processing of the command needs time so that the SDO timeout
would expire with a standard transfer
"

ArthurKetels, 2022-10-21:
Or you can use PDO and have a handshake to guarantee asynchronous state changes. F.e. you define a input object bitfield cmd and an output object bitfield stat. Bit 0 of cmd turns on the LED. Bit 0 of stat shows state of LED (0 = off, 1 = on). The slave checks cmd bit 0 and changes stat depending on the level of bit 0.

nakarlsson, 2022-11-17:
@pauladam316 , can we close this issue?

nakarlsson, 2023-04-29:
close due to inactivity

closed 2023-04-29 by nakarlsson