OpenEtherCATsociety/SOEM #656: Questions about to read/write SDO and PDO?

tom9672, 2022-10-24:
I read the code and run through the slaveinfo and simpletest. I'm new to C and Ethercat, tried to use PYSEOM before, I can use it to do the write/read sdo to config the PDO map, and then make a PDO input and output object to let the output data take my command, also read the input value from servo. Tested on PV,PP,CSP,CSV, and all works as expected.  C and SOEM definitely works better. So, I'm trying to learn this amazing library - SOEM.

Question 1, how to use sdo write? such as I want to write to 0x607F to set the max velocity. Or write to 0x1C12, 0x1C13 to set the customer PDO map.

Question 2, The below code reads the input and output. The output of this code prints ```O: 00 00 00 00 00 00 00 00```, that means I didn't give any output data, How to give output data? such as, if I would like to set the running mode is 1.

Question 3, I got ```I: 00 00 50 02 00 10 e5 b4``` and  ```I: 00 00 50 02 00 00 e5 b4```, what is the meaning of this input? how can I get the current position from this input?

code from `test/linux/simple_test/simple_test.c`
```C
                    if(wkc >= expectedWKC)
                    {
                        printf("Processdata cycle %4d, WKC %d , O:", i, wkc);

                        for(j = 0 ; j < oloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[0].outputs + j));
                        }

                        printf(" I:");
                        for(j = 0 ; j < iloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[0].inputs + j));
                        }
                        printf(" T:%"PRId64"\r",ec_DCtime);
                        needlf = TRUE;
                    }
                    osal_usleep(5000);
```

👍 Thank you so much!

ArthurKetels, 2022-10-25:
Please read the tutorial.txt in the doc folder. It gives examples on exactly the question you ask.

tom9672, 2022-10-25:
Thank you, will check the tutorial.txt later on.

closed 2022-10-25 by tom9672