OpenEtherCATsociety/SOEM #657: Motor is not moving

parinaz6, 2022-10-28:
Hi there,

It is my first time trying to drive a servo with Ethercat communication. Following the provided datasheet I was able to run the motor using Twincat however after spending two weeks I am not able to run the motor with SOEM. Going through SOEM issues and decoding Twincat I was able to do the PDO mapping but unfortunately the motor is not moving and I have no clue what the problem is. There is not enough info for newbies in the datasheet to figure what step I am missing. I have attached the code and datasheet. Can you help me out to figure this?
[eDriver EtherCAT Communication Manual Ver1.5 - EN.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/9892248/eDriver.EtherCAT.Communication.Manual.Ver1.5.-.EN.pdf)
[Uploading MotorControl_SOEM.zip…]()

msevnctkn, 2022-10-29:
the codes are not seeming!

parinaz6, 2022-10-31:
Ok, here is different part of the code:
this is for pdo mapping:
/**
 * \file
 * \ PDO mapping functions for haptic robot
 * 
 * Edited by Zikun Yu -- yu.zik@husky.neu.edu
 *
 * Whitney Robotics Lab  9/5/2018
 */

#include <stdio.h>
#include <stdlib.h>
#include <inttypes.h>
#include <math.h>
#include "ethercat.h"

//uint8  buf8;
//uint16 buf16;
//uint32 buf32;

/*
typedef struct PACKED
{
    int32 ac_pos; // Actual position (6064)
    int32 ac_vel; // Acutal velocity (606C)
} in_motor_t;
PACKED_END

typedef struct PACKED
{
    int32 tg_pos; // Target position (607A)
    int16 tg_tau; // Target torque (6071)
} out_motor_t;
PACKED_END
*/

/** SDO transfer functions 
 *
 * @param[in] slave	=  Slave number.
 * @param[in] index	=  Index to write.
 * @param[in] subindex	=  Subindex to write, must be 0 or 1 if CA is used.	
 * @param[in] value	=  Value to write.
 * @return wkc after a single SDO is trasfered.
 */
int SDO_write8 (uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
    int wkc;
    wkc = ec_SDOwrite (slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

int SDO_write16 (uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
    int wkc;
    wkc = ec_SDOwrite (slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

int SDO_write32 (uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
    int wkc;
    wkc = ec_SDOwrite (slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

/** Selective PDO mapping
 *
 * @param[in] slave	=  Slave number.
 * @param[out] wkc 	=  Working counter.
[... 544 lines cut ...]
                    }
                }
            }
            if(!ec_group[currentgroup].docheckstate)
            	printf("OK : all slaves resumed OPERATIONAL.\n");
        }
        osal_usleep(10000);
    }
}

OSAL_THREAD_FUNC switch_off( void *ptr )
{
    int c; uint16 BUF16;
    (void)ptr;                  /* Not used */

    do
    {
	c = getchar();
	putchar(c);
    }while (c != 'q');

    /* Activate motor drive */
    WRITE_SDO(1, 0x6040, 0, BUF16, 0, "*Control word: motor1*");
    inOP = FALSE;
}

parinaz6, 2022-10-31:
This is the pdo mapping file
/**
 * \file
 * \ PDO mapping functions for haptic robot
 * 
 * Edited by Zikun Yu -- yu.zik@husky.neu.edu
 *
 * Whitney Robotics Lab  9/5/2018
 */

#include <stdio.h>
#include <stdlib.h>
#include <inttypes.h>
#include <math.h>
#include "ethercat.h"

//uint8  buf8;
//uint16 buf16;
//uint32 buf32;

/*
typedef struct PACKED
{
    int32 ac_pos; // Actual position (6064)
    int32 ac_vel; // Acutal velocity (606C)
} in_motor_t;
PACKED_END

typedef struct PACKED
{
    int32 tg_pos; // Target position (607A)
    int16 tg_tau; // Target torque (6071)
} out_motor_t;
PACKED_END
*/

/** SDO transfer functions 
 *
 * @param[in] slave	=  Slave number.
 * @param[in] index	=  Index to write.
 * @param[in] subindex	=  Subindex to write, must be 0 or 1 if CA is used.	
 * @param[in] value	=  Value to write.
 * @return wkc after a single SDO is trasfered.
 */
int SDO_write8 (uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
    int wkc;
    wkc = ec_SDOwrite (slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

int SDO_write16 (uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
    int wkc;
    wkc = ec_SDOwrite (slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

int SDO_write32 (uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
    int wkc;
    wkc = ec_SDOwrite (slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
    return wkc;
}

/** Selective PDO mapping
 *
 * @param[in] slave	=  Slave number.
 * @param[out] wkc 	=  Working counter.
 * */
int pdo_map (uint16 slave)
{
    int wkc = 0;
    
    printf ("Motor drive setup\n");

    wkc += SDO_write8  (slave, 0x1C12, 0, 0);
    wkc += SDO_write8  (slave, 0x1C13, 0, 0);

    /* CSP Inputs */
    wkc += SDO_write8  (slave, 0x1A00, 0, 0);
    wkc += SDO_write32 (slave, 0x1A00, 1, 0x60640020); //Actual position
    
    wkc += SDO_write32 (slave, 0x1A00, 2, 0x606C0020); //Actual velocity
    /* added by me */
    wkc += SDO_write32 (slave, 0x1A00, 3, 0x60770010); //Actual Torque
    wkc += SDO_write32 (slave, 0x1A00, 4, 0x60610008); //Display modes of operation
    wkc += SDO_write32 (slave, 0x1A00, 5, 0x00000008);
    wkc += SDO_write32 (slave, 0x1600, 6, 0x60410010); //status word 
   // wkc += SDO_write16 (slave, 0x1A00, 5, 0x6060);
   // wkc += SDO_write8 (slave, 0x1A00, 5, 0);
    wkc += SDO_write8  (slave, 0x1A00, 0, 6);

    /* CSP Outputs */
    wkc += SDO_write8  (slave, 0x1600, 0, 0);
    wkc += SDO_write32 (slave, 0x1600, 1, 0x60400010); //Control word  
    wkc += SDO_write32 (slave, 0x1600, 2, 0x607A0020); //Target position
   // wkc += SDO_write32 (slave, 0x1600, 2, 0x60710010); //Target torque
    wkc += SDO_write32 (slave, 0x1600, 3, 0x60810020); //Target velocity
    wkc += SDO_write32 (slave, 0x1600, 4, 0x60600008); //Modes of Operation
    wkc += SDO_write32 (slave, 0x1600, 5, 0x00000008);
  //  wkc += SDO_write16 (slave, 0x1600, 4, 0x6060);
   // wkc += SDO_write8 (slave, 0x1600, 5, 0);
    wkc += SDO_write8  (slave, 0x1600, 0, 5);

    wkc += SDO_write16 (slave, 0x1C12, 1, 0X1600);
    wkc += SDO_write8  (slave, 0x1C12, 0, 1);

    wkc += SDO_write16 (slave, 0x1C13, 1, 0x1A00);
    wkc += SDO_write8  (slave, 0x1C13, 0, 1);

    /* Explicitly set flags that are (probably) invalid in EEPROM */
    //ec_slave[slave].SM[2].SMflags = 0x10024l;

    /* Explicitly disable the sync manager that are activated by EEPROM */
    ec_slave[slave].SM[4].StartAddr = 0;
    ec_slave[slave].SM[5].StartAddr = 0;

    return wkc;
}

/** PDO mapping for motor
 *
 * @param[in] slave	=  Slave number.
 * @return -1 if motor setup fails.
 */
int motor_setup (uint16 slave)
{
    ec_slave[slave].PO2SOconfig = pdo_map;

//    if (wkc != 9)
//    {
//	printf ("Motor setup failed!");
//        return -1;
//    }

    return 0; 
}

/** Saturate encoder value within 2*Pi
 *
 * @param[in] encoder	=  Raw value.
 * return saturated value.
 */
double saturation (double encoder)
{
    double val = 2*M_PI; 
    while (fabs(encoder) > val)
    {
	if (encoder > 0) encoder = encoder - 2*M_PI;
	else if(encoder < 0) encoder = encoder + 2*M_PI;
    }
    return encoder;
}

/** Set target position
 * @param[in] slave	=  Slave number
 * @param[in] value	=  Position value
 */
/*
int TargetPosition(uint16 slave, int32 value)
{
    out_motor_t *out_motor = (out_motor_t *)ec_slave[slave].outputs;
    out_motor->tg_pos = value;

    return 0;
}
*/

/** Set target torque
 * @param[in] slave	=  Slave number
 * @param[in] value	=  Torque value
 */
/*
int TargetTorque(uint16 slave, int16 value)
{
    out_motor_t *out_motor = (out_motor_t *)ec_slave[slave].outputs;
    out_motor->tg_tau = value;

    return 0;
}
*/

/** Get actual position
 * @param[in] slave	=  Slave number
 * @param[out] ac_pos	=  Actual position
 */
/*
int GetActualPosition(uint16 slave)
{
    in_motor_t *in_motor = (in_motor_t *)ec_slave[slave].inputs;
    return in_motor->ac_pos;
}
*/

/** Get actual velocity
 * @param[in] slave	=  Slave number
 * @param[out] ac_vel	=  Actual velocity
 */
/*
int GetActualVelocity(uint16 slave)
{
    in_motor_t *in_motor = (in_motor_t *)ec_slave[slave].inputs;
    return in_motor->ac_vel;
}
*/

parinaz6, 2022-10-31:
And this is the sdo config:
/**
 * \file
 * \ Configure motor static parameters.
 * 
 * Edited by Zikun Yu -- yu.zik@husky.neu.edu
 *
 * Whitney Robotics Lab  10/2/2018
 */

#include <stdio.h>
#include <stdlib.h>
#include <inttypes.h>

#include "ethercat.h"
#include "config_map.h"
#include "ecattype.h"

int motor_init(uint16 motor)
{

   // uint8 BUF8;
     uint32 BUF32;

    /* Switch to torque mode */
    /* OpMode: 8 => CSP mode */
    /* OpMode: 10 => Torque mode */
  //  WRITE_SDO(motor, 0x6060, 0, BUF8, 8 , "OpMode");
   // READ_SDO(motor, 0x6061, 0, BUF8, "OpMode display");
    /* rated torque (0.001 Nm) */
    /* 1 Nm = 5000 units (up to 4Nm) */
    /* 1 unit = 0.2 mNm */ 
    WRITE_SDO(motor, 0x6076, 0, BUF32, 30, "Motor rated torque");
    WRITE_SDO(motor, 0x4602, 0, BUF32, 1, "Release Brake");
    WRITE_SDO(motor, 0x6080, 0, BUF32, 314572, "Maximum speed");
  //  WRITE_SDO(motor, 0x60C2, 1, BUF8, 5, "interpolation");
   // WRITE_SDO(motor, 0x60C2, 2, BUF8, 0, "interpolation");
  //  osal_usleep(1000);

    return 0;
}

parinaz6, 2022-10-31:
and here is the config for the state machine:
/**
 * \file
 * \Configuration for running EtherCAT threads.
 *
 * Edited by Zikun Yu -- yu.zik@husky.neu.edu
 *
 * Whitney Robotics Lab  9/10/2018
 */

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"
#include "config_map.h"
#include "ecattype.h"
#include "haptic_config.h"
#include "controller.h"
#include "config_SDOparam.h"

#define EC_TIMEOUTMON 200

char IOmap[4096];
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;
double ActualPosition = 0, ActualVelocity = 0, InputTorque = 0, ReferencePosition = 0, Targetposition=0, Targetvelocity=0, Controlword=0,status=0;

typedef struct PACKED
{
    int32 ac_pos; // Actual position (6064)
    
    int32 ac_vel; // Acutal velocity (606C)
    int16 ac_taw; 
    int8 actual_m; //actual mode
    int16 st_w;//status word 
} in_motor_t;
PACKED_END

typedef struct PACKED
{
    
    int16 ctrl_w; //control word
   
    int32 tg_pos; // Target position (607A)
    //int16 tg_tau; // Target torque (6071)
    int32 tg_vel; 
     int8 op_mode; //operational mode
} out_motor_t;
PACKED_END

void haptic_config(char *ifname)
{
    int i, oloop, iloop, chk; //int j;
    needlf = FALSE;
    inOP = FALSE;
    
   // uint16 BUF16; //uint32 BUF32;
    // uint8 BUF8;  uint32 BUF32;
    uint16 motor1; //uint16 motor2;
	    
    printf("Starting to configure robot\n");

    /* initialise SOEM, bind socket to ifname */
    if (ec_init(ifname))
    {
      	printf("ec_init on %s succeeded.\n",ifname);
[... 312 lines cut ...]
                    }
                }
            }
            if(!ec_group[currentgroup].docheckstate)
            	printf("OK : all slaves resumed OPERATIONAL.\n");
        }
        osal_usleep(10000);
    }
}

OSAL_THREAD_FUNC switch_off( void *ptr )
{
    int c; uint16 BUF16;
    (void)ptr;                  /* Not used */

    do
    {
	c = getchar();
	putchar(c);
    }while (c != 'q');

    /* Activate motor drive */
    WRITE_SDO(1, 0x6040, 0, BUF16, 0, "*Control word: motor1*");
    inOP = FALSE;
}

parinaz6, 2022-10-31:
The motor is in Op state and the PDO values are correct but it is not moving 
[image]

ArthurKetels, 2022-11-05:
You code is not really written to handle state machines very well. Your slave is a DS402 compatible slave, and therefore has DS402 state machine. You write sequential code without even checking for state transitions in the slave. Also CSP mode needs a sync0 signal to be activate and the master to be synchronized to that sync0 pulse. You can read in old posts here how to do that.

nkt17, 2022-12-08:
Hello did you manage to solve this problem. I can launch motor using elmo legacy command (ec_SDOwrite(_slaveNo, 0x3020, 0x01, FALSE, sizeof(int), &p, EC_TIMEOUTSAFE);) and force drive to ignore EC_STATE_SAFE_OP mode by commenting ethercatconfig.c:1303-1307. If drive once entered EC_STATE_SAFE_OP mode it will not start moving and i don't know why. And i don't know how to launch motor via PDO. Can you help me?

paul218, 2023-06-07:
hello I start working with the same servomotor. 
Is someone was able to run the servomotor?