OpenEtherCATsociety/SOEM #658: Using Npcap instead of WinPcap

pdematteis, 2022-11-02:
Since WinPcap is no more supported and deprecated, add support for Npcap.
wpcap.dll should be delayed loaded and C:\Windows\System32\Npcap directory should be added to the search path.
No other changes are needed.

CMakeLists.txt : add linker flags
```
if(WIN32)
  ...
  set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} /DELAYLOAD:wpcap.dll")
```

simple_test.c : add the following code
```
  // Add Npcap dll directory to the search path
   char npcap_dir[MAX_PATH];
   if (GetSystemDirectoryA(npcap_dir, sizeof(npcap_dir)) > 0) {
      strcat_s(npcap_dir, sizeof(npcap_dir), "\\Npcap");
      SetDllDirectory(npcap_dir);
   }
```

ArthurKetels, 2022-11-05:
Seems like a nice addition for windows targets. However the platform is still crappy from a real-time perspective. And also there is the real risk new versions of windows might kill this path to transmit raw ethernet packets.

pdematteis, 2022-11-08:
That's mainly for debug and develop purposes. Regarding Npcap, I think it will survive to Windows restrictions somehow, since I can't imagine a network debug environment without Wireshark :-)