OpenEtherCATsociety/SOEM #660: What is the proper way of setting my motor to operational?

UnderJollyRogers, 2022-11-03:
I am trying to control Qrob motors using SOEM, I use the following code to set the motors to operational mode:
```
static int ConfigureMotors()
    {
        char IOmap[4096];
        static int usedMemory = ec_config_map(&IOmap);
        static int expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
        ec_configdc();
        ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE * 4);
        ec_slave[0].state = EC_STATE_OPERATIONAL;
        ec_send_processdata();
        ec_receive_processdata(EC_TIMEOUTRET);
        ec_writestate(0); /* request OP state for the motor */
        int chk = 200;
        do
        {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            ec_statecheck(0, EC_STATE_OPERATIONAL, 5000);
        } while (--chk && ec_slave[0].state != EC_STATE_OPERATIONAL);
        if (chk == 0){
            return 1;
        }
        else{
            return 0;
        }
        // cout<<"chk: "<<chk<<endl;
    }
```
Although the motors works fine, sometimes the motor refuses to set to OPERATIONAL mode. Let's said that 1 of 4 tries the motors cannot go to OPERATIONAL mode. I know that the hardware is fine because I tested the motors using twincat. But still, I am not sure how to increase the reliability of my code. 

Any suggestion of how can I improve my code of if there is something missing?

ArthurKetels, 2022-11-05:
To have stable and reliable real-time communication with servo slaves see the many older posts about this subject. One I recommend is  https://github.com/OpenEtherCATsociety/SOEM/issues/487#issuecomment-786245585

nakarlsson, 2022-11-17:
@UnderJollyRogers , can this issue be closed?

UnderJollyRogers, 2023-06-19:
Yes, I was able to solve the problem. I did two things,
1. I created a for loop to set each slave in operational mode, one at a time.
For (int i =1; i<=number_of_slaved;i++){
ec_slave[i].state = EC_STATE_OPERATIONAL;
}
2. I avoid using the function: ec_statecheck(0, EC_STATE_OPERATIONAL, 5000); 
Apparently, you just need to check if the motor are in safe operational.

ArthurKetels, 2023-06-19:
Sorry to say this is a fake solution. Setting the local structure to operational does not make the slave go to operational state. The result is simply that the slave stays in safe-OP. 

The fact that you can still control the slave in safe-OP is a violation of the EtherCAT state machine rules. In safe-OP all outputs should remain in a safe state (i.e. motors should be disabled). Ah well, not all slave manufacturers test their product as thorough as mandated I guess.