OpenEtherCATsociety/SOEM #661: Problem with using SOME and ODOT CN-8033

9876ncsystems, 2022-11-16:
Hello, 
We are using successfully SOEM with buscouplers and servo drives from different manufactures, but with ODOT CN-8033 we have problems. 
The traces of simple_test and slaveinfo applications and ESI file are attached. 
[simple_test_CN-8033.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10018928/simple_test_CN-8033.txt)
[slaveinfo_CN-8033.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10018929/slaveinfo_CN-8033.txt)
[XML-ODOT-CN8033-20220630-V1.17.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10018930/XML-ODOT-CN8033-20220630-V1.17.txt)

The device ODOT CN-8033 has 2 modules of digital inputs CT-121F and 2 modules of digital outputs CT-222F.
The only way to goo in OPERATION mode and work was wits direct setting structure fields of the slave:
//<Sm MinSize = "#x22" MaxSize = "#x100" DefaultSize = "#x80" StartAddress = "#x1000" ControlByte = "#x26" Enable = "1">MBoxOut< / Sm>
//<Sm MinSize = "#x22" MaxSize = "#x100" DefaultSize = "#x80" StartAddress = "#x1080" ControlByte = "#x22" Enable = "1">MBoxIn< / Sm>
//<Sm MinSize = "0"    MaxSize = "1024"  DefaultSize = "4"    StartAddress = "#x1100" ControlByte = "#x64" Enable = "1">Outputs< / Sm>
//<Sm MinSize = "0"    MaxSize = "1024"  DefaultSize = "4"    StartAddress = "#x1400" ControlByte = "#x20" Enable = "1">Inputs< / Sm>

         SM[0].StartAddr = 0x1000;
         SM[0].SMlength = 0x0080;
         SM[0].SMflags = 0x00010026;
         SM[1].StartAddr = 0x1080;
         SM[1].SMlength = 0x0080;
         SM[1].SMflags = 0x00010022;

         SM[2].StartAddr = 0x1100;
         SM[2].SMlength = 4;
         SM[2].SMflags = 0x00010064;
         SM[3].StartAddr = 0x1400;
         SM[3].SMlength = 4;
         SM[3].SMflags = 0x00010020;
But any other combination of I/O modules (3, 2, 1, 0) doesn’t works.  
TwinCAT works properly with any combination of I/O modules.
What can advise us?
Regards!

ArthurKetels, 2022-11-16:
The problem is that the slave has a modular device profile but does not implement it fully. For TwinCAT this does not matter as it uses the XML file for the missing information. 

SOEM has to do with the information that is online available in the slave. The CoE object that is missing information is 0x1c12 en 0xc13. They should be filled with the PDO mapping references like 0x1a00 and 0x1a01, 0x1600 and 0x1601.

The result is that SOEM can not read the PDO mapping from the slave and hence does not populate the SM and FMMU structures correctly. Your manual hack works, but it would be better if the slave manufacturer fixed the bug.

9876ncsystems, 2022-11-17:
Thanks!
We discovered that fields of structure SM[0] - SM[3] (except SM[2].SMlength and SM[3].SMlength) are obtained from the device correctly and it is not necessary to set them directly. The device returns for SM[2].SMlength and SM[3].SMlength default value instead of actual one. That is the reason that working combination is only 2 modules of digital inputs and 2 modules of digital outputs. 
We were able under debugger (with direct change of value) to work with combination of 2 modules of digital inputs and 1 module of digital outputs. It seems that other combination should also work.
Regards!

9876ncsystems, 2022-11-19:
Hello!
We made a work around for ODOT CN-8033:
1.	scan the configuration discrete I/O modules with the help of reading parameter 0xf050
2.	direct set the actually values to SM[2].SMlength and SM[3].SMlength 

Thank you for your support!
You can close this issue.

ArthurKetels, 2022-11-19:
Good to know you have a viable solution. However it leaves you with another piece of hardware that needs manual support. Not as good as a proper firmware update that solves the bug.

closed 2022-11-19 by ArthurKetels

gzovermas, 2024-05-23:
I had the same problemc, can you share your code? best regard.