OpenEtherCATsociety/SOEM #662: Soem master using with bewckhoff driver

ahmetihsankose, 2022-11-17:
Hello,
i have a problem with soem master :), i can not reach to safe op of slavess

ahmetihsankose, 2022-11-24:
thank you
I solved this problem according to your post. 
Now, all my slaves ec status is working without error. They are all in OP mode.
But when i tried to read pdo messages i couldn't read anything. Then when I try to run the drivers with the twincat program, I noticed that without installing the Twincat driver manager, these salves not working properly. 
When i install the drive manger, added more settings in Twincat startup settings.  Most of these settings are motor related settings. But, it seems impossible for me to enter all of these settings manually :)

My question is, do I have to enter all these settings for master to work correctly?

The Eni file containing these settings is attached.
Thanks in advance.
[ENI.zip](https://github.com/OpenEtherCATsociety/SOEM/files/10085256/ENI.zip)
hed.

ArthurKetels, 2022-11-24:
You do not have to send all these commands to the slave. Most important are the PreOP to SafeOP transition settings. This means configuring the PDO mapping and also all motor and encoder parameters. Still quite a bit. But this is the way Beckhoff servo slaves are configured. Data is stored in the master and not in the slave. Therefore you have to write a whole bunch of data to each slave before it can operate.

An example :
```
                                                <InitCmd>
							<Transition>PS</Transition>
							<Comment><![CDATA[Ch A Motor parameters / Configured motor type]]></Comment>
							<Timeout>0</Timeout>
							<Ccs>1</Ccs>
							<Index>12992</Index>
							<SubIndex>1</SubIndex>
							<Data>414d383034312d78447830</Data>
						</InitCmd>
```
You have to write to SDO index 0x32c0:01 the data 414d383034312d78447830 (hex array in Complete Access mode). So this is still quite a bit of work. The ideal situation would be if you program your own configurator for this drive. An other solution is to write a ENI parser.

ahmetihsankose, 2022-11-24:
> You do not have to send all these commands to the slave. Most important are the PreOP to SafeOP transition settings. This means configuring the PDO mapping and also all motor and encoder parameters. [...]

Complete access mode will help. 
I don't have the ability and time to write eni file parser for now.
Thank you very much for the answers, I will start communicating with the slaves tomorrow hopefully. :)

ahmetihsankose, 2022-11-30:
Hello, 
ENI parser was not as difficult as I thought. I decided to write it but I have problems with sending data.
I don't fully understand full access mode. According to ENI file , some attributes of init cmds indicate CA = true, but most of init cmds attribute specify nothing like in the example you gave.
// can i write it like this -> 
uint8_t hex_array[11] = {0x41, 0x4d,  0x38, 0x30, 0x,34, 0x31, 0x2d, 0x78, 0x44, 0x78, 0x30};
ec_SDOwrite(slave, 0x32c0, 0x01, TRUE, sizeof(hex_array), &hex_array, EC_TIMEOUTRXM);

When I look at the example below
<InitCmd Fixed="true" CompleteAccess="true">
<Transition>PS</Transition>
<Comment>
<![CDATA[ download pdo 0x1A26 entries ]]>
</Comment>
<Timeout>0</Timeout>
<Ccs>1</Ccs>
<Index>6694</Index>
<SubIndex>0</SubIndex>
<Data>01002000f460</Data>
</InitCmd>

I understand that there is a certain rule abaout Data
Data: 60f4 00 20 0001 (when i look reverse couple data)
Data: Index, subindex, size and length of entry.
But data of 0x32c0:01 means nothing to me.
Maybe my question be a little off topic. Actually i am not a native programmer. i am a very new. 
Can you help abaout this issue, If there is any document abaout this issue, it will suffice as well.
Thanks in advance.

ArthurKetels, 2022-11-30:
All this information you can find in the official ETG document ETG.2100 S (R) V1.0.1 on www.ethercat.org

ArthurKetels, 2023-05-25:
Can you post the wirshark capture (in .pcap format) of the init sequence?

ahmetihsankose, 2023-05-26:
[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11572594/wireshark.zip) this is my wireshark data of the init sequence.

ahmetihsankose, 2023-05-30:
I can keep the slaves in operational mode.
I was continuously calling ecx_send_processdata() and ecx_receive_processdata() in one rt thread. 
When I attempted to read the state of the slaves over EtherCAT from another thread, it seems to disrupt the synchronization.
I guess this synchronization must be something that needs to be provided all over the program.
However i have still problems to keep slaves on op mode for a long time.
Could this still be related to the DC settings? I'm not sure how to determine this through Wireshark.

ahmetihsankose, 2023-06-22:
i have still errors while setting up PDO messages. Specifically, all settings for the first slave power supply is returning a WKC of -5. However, after the initialization is complete, the EtherCAT state remains in error. After that, i use check_ethercat_state function from your example and the slaves can only reach to SafeOp state. Even then, the inputs and outputs in SOEM's slave struct still return nullptr.
[beckhoff_wireshark_data.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11833930/beckhoff_wireshark_data.zip)

closed 2023-12-29 by ahmetihsankose