OpenEtherCATsociety/SOEM #663: Linking against soem. How do we define the includes?

lepalom, 2022-11-21:
There's something that I don't understand, and I think that it is not well-defined. In CMakelist there is section with:

``
target_include_directories(soem PUBLIC
  $<BUILD_INTERFACE:${CMAKE_CURRENT_LIST_DIR}/soem>
  $<INSTALL_INTERFACE:include/soem>)
``

as, _soem_ install in `${CMAKE_PREFIX}/include/soem` if you want to use _soem_, you have to make the includes with:

``
#include <ethercat.h>
``

So, what do you think to change ``$<INSTALL_INTERFACE:include/soem>`` to ``$<INSTALL_INTERFACE:include`` (*maybe not necessary) and use the headers in this way:

``
#include <soem/ethercat.h>
``

ArthurKetels, 2022-11-22:
Great idea, and it does make sense to have the soem/ namespace before any include header files. However, this will also break all existing applications that use SOEM. So no go unfortunately.

lepalom, 2022-11-23:
Well, for that case we have the API compatibility (1.x.x). So, maybe for Soem 2.x.x. it could be possible.