OpenEtherCATsociety/SOEM #664: Question about SEOM Redundancy Mechanism

Lambo0604, 2022-11-23:
I have a question about redundancy about SOEM-1.4.0. Use this interface ec_init_redundant to init two ifs(eth0, eth1) in slaveinfo demo. Howerver，there are some problems:
1.  if two master ports(eth0, eth1) work at the same time, the system works not well.
1.  If link down one master port which connect the last slave OUT port,  the system works well.
2.  If link down one master port which connect the first slave IN port,    the system can only discover the last slave, as if  the last slave not retrans the message to other slaves;
    PS: the slave node has two ports(IN and OUT)
 
The network belows:
  
[image]

   In my opinion, the two master ports send and recv messages at the same time, do I need other operations?  Does the SOEM redundancy have any limits?
   Thanks a lot!

ArthurKetels, 2022-11-24:
Have a look at #621. This is a work-around. I agree there needs to be an update to the configuration functions to handle the redundant case better.

Lambo0604, 2022-11-25:
Thanks for answering.
In my application, I do not care DC sync,  just want to improve communication reliability by the combination of cable redundancy.
in #621, "M1 sends regular packets with PDO, SDO and MAC address A. M2 sends NOP packets with MAC address B. M2 receives packets of M1 and M1 receives packets of M2." 
Does it mean that  I need to custom the soem code, for example : custom the code for cases: M2 sends NOP packets in normal  and sends normal packets in broken happen, only use ec_init_redundant is not enough?

ArthurKetels, 2022-11-25:
The init_redundant function sets all configurations, so usage is transparent for all other SOEM functions. No need to program anything else for redundancy.

What is needed in the application program that you have to supply is some monitoring on the slave states. If you need redundancy it comes natural you want to check and control the failure cases.

Lambo0604, 2022-11-26:
Thanks for answering. But I am still confused by test.
    I use the demo simple_test of Linux for redundant test. I replace ec_init with ec_init_redundant and relace 10000 with 10000000， in order to run a long time. The file simple_test.c has been uploaded.

The network is below:
    master_eth0 -->slave1_IN -->slave1_OUT --> slave2_IN -->slave2_OUT --> master_eth1
 
1. First, link down master eth1, link up eth0,  then run simple_test, the system works well, 2 nodes can be found;

2. Then,  link down master eth0, link up eth1,  the system works fail, 2 nodes can not be found(slave 1 lost;  slave 2 lost);
   at this moment, capture the eth1 message, find that the eth1 sends BRD(offset addr 0) messages continously, but does not recv any message;
   
3. Then,  link down master eth1, link up eth0,  the system work well, 2 nodes can be found;

Now I am very confused. Where  is wrong ?
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095906/simple_test.txt)

Lambo0604, 2022-11-26:
Thanks for answering. But I am still confused by test.

    I use the demo simple_test of Linux for redundant test. I replace ec_init with ec_init_redundant and relace 10000 with 10000000 in order to run a long time. The file simple_test.c has been uploaded.

The network is below:
    master_eth0 -->slave1_IN -->slave1_OUT --> slave2_IN -->slave2_OUT --> master_eth1
 
1. First, link down master eth1, link up eth0,  then run simple_test, the system works well, 2 nodes can be found;

2. Then,  link down master eth0, link up eth1,  the system works fail, 2 nodes can not be found(slave 1 lost;  slave 2 lost);
   at this moment, capture the eth1 message, find that the eth1 sends BRD(offset addr 0) messages continously, but does not recv any message;
   
3. Then,  link down master eth0, link up eth1,  the system work well, 2 nodes can be found;

   Now I am very confused. where is wrong ?

   
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095908/simple_test.txt)

Lambo0604, 2022-11-26:
Thanks for answering. But I am still confused by test.
    I use the demo simple_test of Linux for redundant test. I replace ec_init with ec_init_redundant and relace 10000 with 10000000 in order to run a long time. The file simple_test.c has been uploaded.

    The network is below:
    master_eth0 -->slave1_IN -->slave1_OUT --> slave2_IN -->slave2_OUT --> master_eth1
 
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095909/simple_test.txt)

    1. First, link down master eth1, link up eth0,  then run simple_test, the system works well, 2 nodes can be found;

    2. Then,  link down master eth0, link up eth1,  the system works fail, 2 nodes can not be found(slave 1 lost;  slave 2 lost);
    at this moment, capture the eth1 message, find that the eth1 sends BRD(offset addr 0) messages continously, but does not recv any message;
   
    3. Then,  link down master eth0, link up eth1,  the system work well, 2 nodes can be found;

   Now I am very confused. where is wrong ?

Lambo0604, 2022-11-26:
Thanks for answering. But I am still confused by test.
I use the demo simple_test of Linux for redundant test. I replace ec_init with ec_init_redundant and relace 10000 with 10000000 
in order to run a long time. The file simple_test.c has been uploaded.
The network is below:
master_eth0 -->slave1_IN -->slave1_OUT --> slave2_IN -->slave2_OUT --> master_eth1

1. First, link down master eth1, link up eth0,  then run simple_test, the system works well, 2 nodes can be found;
2. Then,  link down 
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095910/simple_test.txt)
master eth0, link up eth1,  the system works fail, 2 nodes can not be found(slave 1 lost;  slave 2 lost);
    at this moment, capture the eth1 message, find that the eth1 sends BRD(offset addr 0) messages continously, but does not recv any message;
3. Then,  link down master eth0, link up eth1,  the system work well, 2 nodes can be found;
    Now I am very confused. where is wrong ?

Lambo0604, 2022-11-26:
Thanks for answering. But I am still confused by test.
I use the demo simple_test of Linux for redundant test. I replace ec_init with ec_init_redundant and relace 10000 with 10000000
in order to run a long time. The file simple_test.c has been uploaded.
The network is below:
master_eth0 -->slave1_IN -->slave1_OUT --> slave2_IN -->slave2_OUT --> master_eth1

1）First, link down master eth1, link up eth0, then run simple_test, the system works well, 2 nodes can be found;
2）Then, link down
 master eth0, link up eth1, the system works fail, 2 nodes can not be found(slave 1 lost; slave 2 lost);
at this moment, capture the eth1 message, find that the eth1 sends BRD(offset addr 0) messages continously, but does not recv any message;
3）At last, link down master eth0, link up eth1, the system work well, 2 nodes can be found;
Now I am very confused. where is wrong ?

[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095911/simple_test.txt)

Lambo0604, 2022-11-29:
I use the demo simple_test of Linux for redundant test. I replace ec_init with ec_init_redundant and relace 10000 with 10000000 in order to run a long time. The simple_test.c is uploaded.
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10111959/simple_test.txt)

Master_eth0 -->slave1_IN -->slave1_OUT --> slave2_IN -->slave2_OUT --> Master_eth1
 
1. First, link down master eth1, link up eth0,  then run simple_test, the system work well, 2 nodes can be found;
2. Then,  link down master eth0, link up eth1,  Only slave2 can be find.
3. Then,  link up master eth0 and eth1,  only slave2 can be find, slave1 state is 0;

In my opinion, the above three cases should work correctly(2 nodes be found); But the test result is not the same;
pls give some suggestions, thanks very much.

closed 2022-12-08 by Lambo0604