OpenEtherCATsociety/SOEM #665: PDOmapping with microflex e190

s2cihero, 2022-11-23:
Hey team,

just i need answer for basic question.

i can read and write SDO,but i need the control servo.i need PDO,but i dont understand 1600 index mapping.anyone help me for this ?

drive xml file 1600 index;

<RxPdo Fixed="0" Mandatory="1" Sm="2">
<Index>#x1600</Index>
<Name>Outputs</Name>
<Entry>
<Index>#x6040</Index>
<SubIndex>0</SubIndex>
<BitLen>16</BitLen>
<Name>AX0_ControlWord_U16</Name>
<DataType>UINT</DataType>
</Entry>
<Entry>
<Index>#x607A</Index>
<SubIndex>0</SubIndex>
<BitLen>32</BitLen>
<Name>AX0_TargetPosition_I32</Name>
<DataType>DINT</DataType>
</Entry>
</RxPdo>

1600 index have 256 subindex,any default table ? or how can i assign this ?

s2cihero, 2022-11-25:
hello again,
i already search but no answer my questions.

if (ec_init(ifname))
    {
        printf("ec_init on %s succeeded.\n", ifname);
        if (ec_config(TRUE, &IOmap) > 0)
        {
        ...
uint8 OPmode = 8;
              int16 isub6040 = 6;
           uint16 iisub6040 = 7;
           uint16 iiisub6040 = 15;
            
            ec_dcsync0(1, TRUE,20000 , 150000);
            ec_SDOwrite(1, 0x6060, 0x00, FALSE, 2, &OPmode, EC_TIMEOUTSAFE);
             ec_SDOwrite(1, 0x6040, 0x00, FALSE, 2, &isub6040, EC_TIMEOUTSAFE);
             ec_SDOwrite(1, 0x6040, 0x00, FALSE, 2, &iisub6040, EC_TIMEOUTSAFE);
             ec_SDOwrite(1, 0x6040, 0x00, FALSE, 2,  &iiisub6040, EC_TIMEOUTSAFE);

1- ) my code for control word there,but sync status does not change to SYNC0.where is my fault ?

2-) i dont understand PDO mapping , i have been trying to understand this for 3 weeks.but no solution. 
2-a) how can i map for servo control or axis move ? any example for this ? (tutorial.txt not good to explain.)

im adding drive xml file.

[ABB MicroFlex e190 Build 5904.6 (CoE).txt](https://github.com/OpenEtherCATsociety/SOEM/files/10088225/ABB.MicroFlex.e190.Build.5904.6.CoE.txt)

anyone can help me ?

nakarlsson, 2023-04-29:
@mithathacioglu , can we close this issue?

mithathacioglu, 2023-04-30:
Hello mate,Yeah u can close the issue,no solution but i leaved my job :D

closed 2023-05-01 by nakarlsson