OpenEtherCATsociety/SOEM #666: Cannot transition to Operational (Elmo Platinum motor controller)

d-loret, 2022-11-24:
I am not sure where the issue lies, but our Elmo Platinum motor controller cannot transition to Operational through SOEM's `ecx_writestate`. It gets stuck in SAFEOP, Error.
The transition is attempted after the PDO mapping is configured. As far as I can see, the PDO mapping is configured successfully. One way I confirmed this is with slaveinfo (output at the end of this post).

Slaveinfo output:
```
Slaveinfo
Starting slaveinfo
ec_init on enp89s0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:ModuleSlotsDrive
 Output size: 232bits
 Input size: 296bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000009a ID: 01100002 Rev: 00120027
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  29 F:00010064 Type:3
 SM3 A:1c00 L:  37 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  29 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001d Ll:  37 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0004.0] 0x60FF:0x00 0x20 INTEGER32    Target velocity
  [0x0008.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  [0x000A.0] 0x60B0:0x00 0x20 INTEGER32    Position offset
  [0x000E.0] 0x60B1:0x00 0x20 INTEGER32    Velocity offset
  [0x0012.0] 0x60B2:0x00 0x10 INTEGER16    Torque offset
  [0x0014.0] 0x6060:0x00 0x08 INTEGER8     Modes of operation
  [0x0015.0] 0x6073:0x00 0x10 UNSIGNED16   Max current
  [0x0017.0] 0x60FE:0x01 0x20 UNSIGNED32   SubIndex 001
  [0x001B.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x001D.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x0021.0] 0x6069:0x00 0x20 INTEGER32    Velocity sensor actual value
  [0x0025.0] 0x6078:0x00 0x10 INTEGER16    Current actual value
  [0x0027.0] 0x6061:0x00 0x08 INTEGER8     Modes of operation display
  [0x0028.0] 0x6079:0x00 0x20 UNSIGNED32   DC link circuit voltage
  [0x002C.0] 0x3610:0x00 0x20 REAL32       Drive temperature in celsius
  [0x0030.0] 0x60FD:0x00 0x20 UNSIGNED32   Digital inputs
  [0x0034.0] 0x2205:0x01 0x10 INTEGER16    SubIndex 001
  [0x0036.0] 0x2205:0x02 0x10 INTEGER16    SubIndex 002
  [0x0038.0] 0x3607:0x01 0x20 UNSIGNED32   SubIndex 001
  [0x003C.0] 0x3607:0x02 0x20 UNSIGNED32   SubIndex 002
  [0x0040.0] 0x6041:0x00 0x10 UNSIGNED16   Statusword
End slaveinfo, close socket

d-loret, 2022-11-24:
Wireshark log:
[debug_safeop_to_op_error.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10082199/debug_safeop_to_op_error.pcapng.txt)

ArthurKetels, 2022-11-24:
Use the search function on this forum.
"is:issue platinum"
The slave has some peculiar hardware from TI that need special care. You need to block LRW and use overlapped mode.

d-loret, 2022-11-24:
Thanks for your reply.

Do I need both or one or the other? If I only set `blockLRW` to 1, or if I only change `ecx_config_map_group` and `ecx_send_processdata` to `ecx_config_overlap_map_group` and `ecx_send_overlap_processdata`, respectively, the drive does get into OPERATIONAL. However if I block LRW AND use the overlap functions, it still gets stuck in SAFEOP.

If we were using the legacy mapping (i.e. non-overlap functions) for any device up to now, should we create a separate group for the Platinum drives? Or do most devices out there support overlapping too? I ask this in case we need the overlap functions (and not setting `blockLRW`).

As a side note, once in OPERATIONAL, I cannot enable the servo, but I need to debug that more.

d-loret, 2022-12-09:
@ArthurKetels, is the use of `blockLRW` and `ecx_config_overlap_map_group`/`ecx_send_overlap_processdata` mutually exclusive? Is it expected that when trying to use both, OPERATIONAL cannot be reached? It can only be reached when using only one.

Also, what is the difference between setting `blockLRW` to 1 vs using the overlap functions?

Even though the EtherCAT state machine can now transition to operational, the drive is not responding to commands in the RxPDO, so I feel I am missing something.

nakarlsson, 2022-12-09:
I think you can use either.   block LRW  OR overlapped mode

For info on the root casue, 3.6.3 Fix of PINDSW-141 Errata by the SOEM Master and IgH Master
https://www.ti.com/lit/an/spracj7/spracj7.pdf?ts=1670570597811&ref_url=https%253A%252F%252Fwww.google.com%252F