OpenEtherCATsociety/SOEM #667: Can not go to OP State with SOEM using XMC4800

yongzhiliu1006, 2022-11-26:
Hi :
    I am using XMC4800 of infineon for ethercat slave controller for a project, and generate slave stack files with SSC TOOL 5.12.  The problem I met  is that when the amount is more than 4 or 6, the slaves are more difficult to transit to OP state,using SOEM''s simple test. I checked the captured packets with wireshark, the master found all the 11 slaves;
  ................................................................
  . after read from 0x130 AL status regs,(the slaves are all in preop state) ;
  . the soem master read the regs of 0x5XX from the second slave  (adp 0x1002) to the laset slave. 
  . the soem master read fro
[pcap.zip](https://github.com/OpenEtherCATsociety/SOEM/files/10095869/pcap.zip)
m the first slave (adp 0x1001) of the 0x9xx regs, just finished the visit of 0x1001 and 0x1002;
 . then bwr all slaves's Ado 0x120 regs.
[pcap.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095872/pcap.txt)
[XMC_ESC.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10095873/XMC_ESC.txt)

 . then the comunication between seem to be lost .
I attached the wireshark packets file and the esi file below. Both of the file is renamed to txt to be uploaded.
Any help and guidance would be greatly appreciated! 
Thank you so much.

ArthurKetels, 2022-11-27:
Can you show the output of slaveinfo -map?

I can not see the useless FPRD you mention in the pcap file. It does query the slave status for a long time to see if slaves go to operational.

You mention the slaves return with a AL status error. This is very important because it tells you why the slave refused to go in safe-op. The AL status code for all slaves is 0x001e. This means "Invalid input configuration". Because I see no CoE mailbox data I assume you configured the slave to have EEPROM interface only. This in turn means that your EEPROM PDO configuration does not match the slave expectations when transitioning from pre-op to safe-op.

yongzhiliu1006, 2022-11-27:
Dear sir:

yongzhiliu1006, 2022-11-27:
Dear sir:
   Thanks a lot for your advice. The word "useless" is not a proper usage. I reviewed the communication packet between master and slaves,found many packets with Ado 0x00 in INIT state. The slaveinfo -map result is posted below. 
   
[image]

ArthurKetels, 2022-11-27:
As far as I can see this is a slave firmware issue and not something SOEM related. Did you test the slave with the slave conformance test tool from the ETG? It is very good at finding firmware issues with any slave.

yongzhiliu1006, 2022-11-27:
Sorry sir, I am not a member of ETG, and donot have a Conformance Test Tool by my hand. But I tried TwinCat ,the slaves can be connected quite easily. Where can I get any clue about what kind of  issue it can be with the firmware? Thanks again.

ArthurKetels, 2022-11-28:
Well, if you do not want to use the proper methods and proper tools then it will be more of a struggle. One thing that could help is posting a complete set of data instead of small snippets. 

The post of slaveinfo -map is not complete (shows only one slave and does not have the '-map' function included). Also the corresponding .pcap file would be useful.

I suspect some mistakes in the eeprom content. Can you post the binary result of eepromtool? I have software to crosscheck the eeprom binary with the .esi.

yongzhiliu1006, 2022-11-29:
Thank you ,sir. 
The 11 slavesinfo -map result and the e2prom file are packed.
[Pictures.zip](https://github.com/OpenEtherCATsociety/SOEM/files/10109652/Pictures.zip)