OpenEtherCATsociety/SOEM #668: beckoff EK1100 pdo read

s2cihero, 2022-11-29:
hello guys,

just i need a basic quesiton.

 

this screenshot taken in twincat3, this device have default PDO map,

[image]
[image]

but i cannot read this pdo data witth this code, how can i read ?

```
uint8_t bit1 = 1;
int bit =1 ;
ec_SDOread(2, 0x6000, 0x0e, FALSE, &bit, &bit1, EC_TIMEOUTSAFE);
```

i tried 1c10 and 1a00 registers for read. same problem.

ArthurKetels, 2022-11-29:
See tutorial.txt in the /doc folder. This is a slave without any microcontroller, so no mailbox and no CoE communication. ec_SDOread() uses CoE. You need to manipulate the data in the IOmap memory array with PDO transfers.

Edphoenixmax, 2023-03-27:
may I ask why the second parameter is 0x6000?

ArthurKetels, 2023-03-27:
From the source code:
```
/** CoE SDO read, blocking. Single subindex or Complete Access.
 *
 * Only a "normal" upload request is issued. If the requested parameter is <= 4bytes
 * then a "expedited" response is returned, otherwise a "normal" response. If a "normal"
 * response is larger than the mailbox size then the response is segmented. The function
 * will combine all segments and copy them to the parameter buffer.
 *
 * @param[in]  slave      = Slave number
 * @param[in]  index      = Index to read
 * @param[in]  subindex   = Subindex to read, must be 0 or 1 if CA is used.
 * @param[in]  CA         = FALSE = single subindex. TRUE = Complete Access, all subindexes read.
 * @param[in,out] psize   = Size in bytes of parameter buffer, returns bytes read from SDO.
 * @param[out] p          = Pointer to parameter buffer
 * @param[in]  timeout    = Timeout in us, standard is EC_TIMEOUTRXM
 * @return Workcounter from last slave response
 * @see ecx_SDOread
 */
int ec_SDOread(uint16 slave, uint16 index, uint8 subindex,
               boolean CA, int *psize, void *p, int timeout)
```
The second paramter is de CoE index, the third is the subindex. This combination uniquely identifies the CoE paramter from the slave. In the case above 6000:01 is the first PDO mapped parameter of that slave. But again, this fails for that slave because it has no CoE functionality.

closed 2023-03-28 by ArthurKetels