OpenEtherCATsociety/SOEM #672: Hello,I have a strange question

965295342, 2022-12-08:
My other program A will provide data to soem program, and then send the data to EL6751. Meanwhile, program A will communicate with program B through UDP. When I connect to EL6751 to run program A, the UDP of my program A seems to be invalid. I couldn't communicate with program B by running program A for several hours. But after A few hours, my linux computer would return to normal, and I could run program A and Program B to communicate using UDP. As A result, whenever I use soem and connect to EL6751, my program A cannot UDP communicate with program B for several hours

ArthurKetels, 2022-12-08:
Perhaps a routing issue? Your EtherCAT port should not be part of any network routing.

965295342, 2022-12-08:
> Perhaps a routing issue? Your EtherCAT port should not be part of any network routing.

Hello, I use the translation software to speak English. The ports used by my two programs are 24000 and 25000. At first, I thought that these two ports were used by soem. Do you mean that soem does not use these two ports? Strangely enough, A few hours after the soem program caused the exception, my program A and B were able to communicate using UDP. I guess that's why I didn't use ec_close()? My soem program may be blocking my UDP communication. I don't know if my conjecture is reasonable. Thank you for your patience.

closed 2023-02-03 by 965295342