OpenEtherCATsociety/SOEM #673: Optimizing controlling drive DC402 by velocity(position)

nkt17, 2022-12-11:
Hello everyone, i'm not expert with EtherCAT.

I've managed to launch drive by executing Elmo legacy commands JV(PA) and BG (e. g. JV=111111: ec_SDOwrite(slave, 0x30ff, 0x01, FALSE, 4, 111111, TIMEOUTSAFE), BG: ec_SDOwrite(slave, 0x3020, 0x01, FALSE, 2, 0x0000, TIMEOUTSAFE). This 2 commands spend approximately 20ms. Then i can't set velocity more than 50 times per second, but i want set velocity(position) around 500 times per second for more precise motor controlling by velocity(position). Can I execute velocity setting command (like JV) and launching motor (like BG) faster (around 1-2 ms for these 2 commands)? 
I've heard that PDO is mush more faster than SDO. I've found function (int ecx_RxPDO(ecx_contextt *context, uint16 Slave, uint16 RxPDOnumber , int psize, const void *p);) but i don't understand what is RxPDOnumber. My drive is DC402.

I will be grateful for your help.

nkt17, 2022-12-11:
Also i've tried to connect to drives via TwinCAT. I've pushed to scan button but there is no devices were found.

ArthurKetels, 2022-12-11:
Perhaps a read of tutorial.txt is in order. Then have a look how simple_slave (from the test programs) works then have a look here on dozens of post that show how to do that. Also the ETG is you friend when you want to gather more knowledge about how EtherCAT actually works. SOEM is a library that assumes the user knows about basic EtherCAT things.

nkt17, 2022-12-14:
Hello again.

I've made custom mapping PDO and tried to send pdo commands to slave. I notices that target velocity (0x60ff) value changed but motor didn't start moving, although doing the same sequence of commands via SDO made motor moving. I guess that happens because the motor for some reasons falls in SAFE_OP mode while sending process_data. Here is my modified simple_test code:

/** \file
 * \brief Example code for Simple Open EtherCAT master
 *
 * Usage : simple_test [ifname1]
 * ifname is NIC interface, f.e. eth0
 *
 * This is a minimal test.
 *
 * (c)Arthur Ketels 2010 - 2011
 */

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

#define pe printError()

PACKED_BEGIN
typedef struct PACKED
{
    int32 tg_vel; // Target velocity (60ff)
    uint16 ctrl_w; // Control word (6040)
    int8 op; // modes of operation (6060)
} out_motor_t;
PACKED_END

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

void printError()
{
    ec_errort err;
    while (ec_poperror(&err))
    {
        printf("Slave %d error!!! AbortCode is %8.8X\n", 1, err.AbortCode);
        printf("\t%s", ecx_err2string(err));
        int32_t p;
        int psize = sizeof(p);
        ec_SDOread(1, 0x306a, 0x01, FALSE, &psize, &p, EC_TIMEOUTSAFE); //EC
        printf("\t\tError code: %d\n", p);
    }
}

void SDOchDebug8u(uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
    printf("_______________________________________________________________________________________________\n");

    uint8 p = 0x88;
    int psize = sizeof(p);
    ec_SDOread(slave, index, subindex, FALSE, &psize, &p, EC_TIMEOUTSAFE);
    printf("0x%4.4X before setting: 0x%2.2X\n", index, p);

    printf("setting 0x%4.4X: %d\n", index, value);
    uint8 u8val = value;
    ec_SDOwrite(slave, index, subindex, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTSAFE);

    p = 0x88;
[... 432 lines cut ...]
   {
      /* create thread to handle slave error handling in OP */
//      pthread_create( &thread1, NULL, (void *) &ecatcheck, (void*) &ctime);
      osal_thread_create(&thread1, 128000, &ecatcheck, (void*) &ctime);
      /* start cyclic part */
      simpletest(argv[1]);
   }
   else
   {
      ec_adaptert * adapter = NULL;
      printf("Usage: simple_test ifname1\nifname = eth0 for example\n");

      printf ("\nAvailable adapters:\n");
      adapter = ec_find_adapters ();
      while (adapter != NULL)
      {
         printf ("    - %s  (%s)\n", adapter->name, adapter->desc);
         adapter = adapter->next;
      }
      ec_free_adapters(adapter);
   }

   printf("End program\n");
   return (0);
}

nkt17, 2022-12-14:
Here is how PDO mapping looks like in slave_info.c
[image]

nkt17, 2022-12-14:
there is how status params look like before and after PDO commands
[image]
[image]

closed 2022-12-19 by nkt17