OpenEtherCATsociety/SOEM #674: How to read lost-link (x0310) and frame error (x030*) counters

DexTrewitt, 2022-12-22:
We have some large EtherCAT chains and want to keep track of low-level errors and, especially, do diagnosis if the PDO exchange fails.   I learned about these from [EtherCAT Diagnosis For Users](https://www.ethercat.org/download/documents/EtherCAT_Diagnosis_For_Users.pdf#page=15&zoom=100,0,0), which has a lot of useful information.

These are generally supported at the very lowest level of the chipset firmware.  In any case, I can't read them with the usual SDO read operations.  Is there some other way to read them?

ArthurKetels, 2022-12-22:
You can use FPRD to directly read registers in a slave ESC.

```
/** FPRD "configured address read" primitive. Blocking.
 *
 * @param[in] port        = port context struct
 * @param[in]  ADP        = Address Position, slave that has address reads.
 * @param[in]  ADO        = Address Offset, slave memory address
 * @param[in]  length     = length of databuffer
 * @param[out] data       = databuffer to put slave data in
 * @param[in]  timeout    = timeout in us, standard is EC_TIMEOUTRET
 * @return Workcounter or EC_NOFRAME
 */
int ecx_FPRD(ecx_portt *port, uint16 ADP, uint16 ADO, uint16 length, void *data, int timeout)
```

For example:
`wkc = ec_FPRD(ec_slave[2].configadr, 0x0310, sizeof(buf), &buf, EC_TIMEOUTRET);`

This reads 0x0310 for length of buf from slave 2. Because minimum frame length of ethernet  is 64 bytes you can as well read all registers of error counters in one go. The most straightforward solution is to use an appropriate struct.

DexTrewitt, 2023-01-01:
This worked perfectly!  Thanks very much.

closed 2023-01-01 by DexTrewitt