OpenEtherCATsociety/SOEM #675: Failed to write xml to eeprom

yanzhang920817, 2023-02-07:
I write the bin format xml to the eeprom of the slave station through the foe method, and the function returns successfully. However, after restarting, it occasionally appears that the xml is still the old version. Who can help me? thanks.

ArthurKetels, 2023-02-07:
The eeprom can not be programmed via FoE. You can find the special tool for this in test/linux/eepromtool . It can read/write bin and hex files to and from eeprom.

yanzhang920817, 2023-02-07:
sorry，eepromtool is not FOE？Now，my question is using eepromtool write xml failed occasionally.

ArthurKetels, 2023-02-07:
There are multiple reasons. Some slaves do not allow the eeprom to be changed. Other slaves only emulate eeprom. And again others have very slow programming, leading to timeouts. You can use the eepromtool to read too. It is trivial to mash up a verify after write to make sure the content has changed.

yanzhang920817, 2023-02-08:
Using eepromtool, I can probably write xml to the slave eeprom, and I can also see that it has been updated after restarting, but there is a small probability that the xml has not been updated after restarting.