OpenEtherCATsociety/SOEM #676: NI 9144 - Error 0x0017 : Invalid sync manager configuration

cnicho35, 2023-02-07:
Hello,

I am trying to read IO from NI modules that are connected to a National Instruments 9144 chassis. I am trying to run the simple_test.c on Linux with the PO2SO function below.

Background: the NI 9144 has a 9239 (4x analog inputs) and a 9401 (8x DIO) mounted in it. Using NI's xml creator tool, I made the xml file that is attached below. I want to have a PDO for the 4 analog inputs (from slave to master) and the 8 digital outputs (from master to slave).

The simple test output is:
```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp3s0 succeeded.
1 slaves found and configured.
found a 9144 
ni_9144 slave 1 set, retval = 5
Slaves mapped, state to SAFE_OP.
segments : 1 : 14 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x0017 : Invalid sync manager configuration
Request init state for all slaves
End simple test, close socket
End program
```
PO2SO Function:
`ec_slave[1].PO2SOconfig = ni_slave;`
```
int ni_slave(uint16 slave)
{
   int retval;

   // map velocity
   // uint16 map_1c12[1] = {0x1601};
   // uint16 map_1c13[2] = {0x1a00, 0x1a01};

   retval = 0;
   uint8 u8val;
   uint16 u16val;

   u8val = 0x01;
   retval += ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
   u16val = 0x1601;
   retval += ec_SDOwrite(slave, 0x1c12, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);

   u8val = 0x02;
   retval += ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
   u16val = 0x1a00;
   retval += ec_SDOwrite(slave, 0x1c13, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
   u16val = 0x1a01;

   retval += ec_SDOwrite(slave, 0x1c13, 0x02, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);

   printf("ECATERROR: %s \n", ec_elist2string());

   printf("ni_9144 slave %d set, retval = %d\n", slave, retval);
   return 1;
}
```

The slave info is as follows:
```
Slave:1
 Name:? M:000001f9 I:000023b8
 Output size: 8bits
 Input size: 104bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
[... 265 lines cut ...]
**Can someone please help me determine why I am getting this SyncManager error?**

From Beckhoff:
Error Code 0x0017
Meaning

Invalid Sync Manager configuration
Description

Process data communication (cyclic communication) is done via extra memory areas on the ESC, separated for outputs and inputs. The process data length and the process data SyncManager length have to be the same. If this is not the case or the start address or direction does not match this error is returned.

Example: The process data configuration was changed of the slaves which also changed the length of the data. The change was not activated in the configuration so that the configuration tool would have recalculated the SyncManager settings.
Current State (or state change)

P→S, S→O
Resulting state

Current State + E
Solution

Issue a re-calculation of the EtherCAT configuration

[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10679699/simple_test.txt)

NI 9144 XML File: [9239_9401_Scan_xml.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10679619/9239_9401_Scan_xml.txt)

ArthurKetels, 2023-02-10:
There is something wrong with the information the slave returns to SOEM. Most critical are the SM memory addresses. According to your XML it should be:
SM0 : 0x1f00
SM1 : 0x1c00
SM2 : 0x1000
SM3: 0x1600

The addresses SOEM gets from the slave are:
SM0 :  0x1c00
SM1 : 0x1800
SM2 : 0x0000
SM3 : 0x0000

This will cause the slave to complain with error code 0x0017.  Either ask for a slave firmware update to fix the bug or hack the SM settings yourself in the ni_slave() function. Although SM0 and SM1 are off, they do seem to work. So you could be lucky and get  away with redefining SM2, SM3, FMMU0 and FMMU1. 

You also made a small mistake. The procedure to remap a slave is to first set subindex 0 to value 0 then update the mapping entries, and last set subindex 0 to the correct value.

rtzoeller, 2025-01-22:
@cnicho35 were you able to get this working?

cnicho35, 2025-01-22:
@rtzoeller I ended up abandoning ship on this effort. It seems easier to just use the Beckhoff modules for IO rather than this NI 9144/9145 chassis. If you had some requirement for the 9144/9145, I bet you could get it working without too much hassle.
if so, good luck.