OpenEtherCATsociety/SOEM #677: Fails to go into SAFEOP Mode

jvbpv, 2023-02-09:
Hello, I am attempting to connect and communicate to a turbo pump over ethercat. I have successfully interfaced with a pressure control valve, mass flow conrtroller and a capacitance diaphram gauge using the python wrapper for this library with no issues. I have now ran into an issue that I have attempted to debug for a few days now. 

When running the slaveinfo program from this library I get the following:

``` .log 
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s8 succeeded.
Time:1675908393.273 SDO slave:1 index:1c12.00 error:08000021 Data cannot be transferred or stored to the application because of local control
Time:1675908393.287 SDO slave:1 index:1c13.00 error:08000021 Data cannot be transferred or stored to the application because of local control
Time:1675908393.370 SDO slave:1 index:0001.00 error:06020000 The object does not exist in the object directory
Time:1675908393.406 SDO slave:1 index:b981.00 error:06020000 The object does not exist in the object directory
1 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:netX
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000005e9 ID: 10000000 Rev: 01010002
 SM0 A:1000 L: 128 F:00010036 Type:1
 SM1 A:1080 L: 128 F:00010032 Type:2
 SM2 A:1100 L:   1 F:00010074 Type:3
 SM3 A:1d00 L:   7 F:00010030 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program
``` 

When I run the ./slaveifo -sdo program I get:
[slaveinfo_pdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/10693212/slaveinfo_pdo.txt)

I have attempted configuring the SM2 and SM3 with default and user set Pdos but both implementations fail upon config_map. I attempted this solution with no success #311 

Upon doing some debugging, I see that the device does not successfully go into SAFE_OP mode, which relates to this error I have come across:
```.error
Time:1675908393.273 SDO slave:1 index:1c12.00 error:08000021 Data cannot be transferred or stored to the application because of local control
Time:1675908393.287 SDO slave:1 index:1c13.00 error:08000021 Data cannot be transferred or stored to the application because of local control
```

Is there an issue with the device ethercat firmware that SOEM isn't working or is there some extra configurations I need to take into account?

ArthurKetels, 2023-02-10:
The output of the object dictionary is like this device is still in very early alpha phase. Most entries are unfinished and often contain garbage. A manufacturer device name of 'XXXX' is a actually a giveaway. Please ask for a slave firmware update.

jvbpv, 2023-02-10:
Hi @ArthurKetels,

I apologize for that, I modified some of the identity data in the text file for privacy reasons.
This device has the latest firmware update.

ArthurKetels, 2023-02-11:
Even with the correction my original point stands. The Object Dictionary and EEPROM data of this slave are not correct. It will not pass the mandatory slave conformance test. Probably this slave has only been tested with masters that configure the slave via an offline configuration tool. The tool takes the slave ESI file and generates an ENI file that is used to program the SM en FMMU. This way the internal data source of the slave (EEPROM and SDO) are omitted. However it is mandatory for each slave manufacturer to provide both methods and test them with the CTT (conformance test tool).