OpenEtherCATsociety/SOEM #680: When bit3 is 1, is the mailbox available or full?

yanzhang920817, 2023-02-28:
When bit3 is 1, is the mailbox available or full?
In the following code, bit3=1 means that the mailbox can be used, but in the screenshot below, bit3=1 means that the mailbox is full.

`int ecx_mbxreceive(ecx_contextt *context, uint16 slave, ec_mbxbuft *mbx, int timeout)
{
   uint16 mbxro, mbxl, configadr;
   int wkc = 0;
   int wkc2;
   uint16 SMstat;
   uint8 SMcontr;
   ec_mbxheadert *mbxh;
   ec_emcyt *EMp;
   ec_mbxerrort *MBXEp;

   configadr = context->slavelist[slave].configadr;
   mbxl = context->slavelist[slave].mbx_rl;
   if ((mbxl > 0) && (mbxl <= EC_MAXMBX))
   {
      osal_timert timer;

      osal_timer_start(&timer, timeout);
      wkc = 0;
      do /* wait for read mailbox available */
      {
         SMstat = 0;
         wkc = ecx_FPRD(context->port, configadr, ECT_REG_SM1STAT, sizeof(SMstat), &SMstat, 3*EC_TIMEOUTRET);
         SMstat = etohs(SMstat);
         if (((SMstat & 0x08) == 0) && (timeout > EC_LOCALDELAY))
         {
            osal_usleep(EC_LOCALDELAY);
         }
      } while (((wkc <= 0) || ((SMstat & 0x08) == 0)) && (osal_timer_is_expired(&timer) == FALSE));

      if ((wkc > 0) && ((SMstat & 0x08) > 0)) /* read mailbox available ? */   (**Here is my question**)
      {
         mbxro = context->slavelist[slave].mbx_ro;
         mbxh = (ec_mbxheadert *)mbx;
         do
         {`

[image]

ArthurKetels, 2023-03-01:
Yes, a mailbox must be full before the ecx_mbxreceive() function can -read- its content. Seems normal logic to me. Just like a mailbox mist be empty before you can -write- something in it.

yanzhang920817, 2023-03-01:
I don't quite understand why I have to wait for the mailbox to be full before I read it. In the screenshot below, I sent a FOE request to the slave station, but why does the slave keep returning to 00C3? Is the timeout too short?

[image]

yanzhang920817, 2023-03-01:
When I send a FOE request to the slave , does the slave station set this mailbox to full，or the soem  set it?

yanzhang920817, 2023-03-02:
The same slave firmware, the same Foe tool, and the same slave location of different devices, but some devices can't write in the slave. After the SOEM sends a write request,  the **ecx_mbxreceive()** function always returns - 5. Use wireshark to capture the packet, and you can see that it has been read to 0x00c3. What is the reason?

`   wkc = ecx_mbxsend(context, slave, (ec_mbxbuft *)&MbxOut, EC_TIMEOUTTXM);
   if (wkc > 0) /* succeeded to place mailbox in slave ? */
   {
      do
      {
         worktodo = FALSE;
         /* clean mailboxbuffer */
         ec_clearmbx(&MbxIn);
         /* read slave response */
         wkc = ecx_mbxreceive(context, slave, (ec_mbxbuft *)&MbxIn, timeout);
         if (wkc > 0) /* succeeded to read slave response ? */
         {`

ArthurKetels, 2023-03-02:
I do not understand your question. SOEM is behaving as it should. No bug here. The slave behaves strange, but not outside the protocol specifications. If you have issues with your slave, ask the slave manufacturer. If you need larger time-out values, simply modify SOEM to your liking.  I have seen slaves that take up to 20 seconds to respond to a request.

closed 2023-03-03 by yanzhang920817