OpenEtherCATsociety/SOEM #681: Can PDO data be read and then sent?

lhf739181583, 2023-02-28:
I wanted to use the master to read the data before sending it, but I found that the LRW was only available when run ec_send_processdata(). I tried to separate the LRD,LWR in ecx_main_send_processdata , but the result was not satisfactory

ArthurKetels, 2023-03-01:
LRD, LWR, LRW are available for use in an application. But be aware that the conditions to use it need to be set-up correctly. All of these methods use logical addressing. And need the proper set-up of FMMU's in each slave so as to set the correct physical to logical mapping. If the result is not satisfactory then clearly you have not set the correct conditions.

lhf739181583, 2023-03-01:
Thank you very much for your answer. I am curious why it is necessary to read when sending instead of reading before sending. Because in general, we need to obtain the information of the slave station first, after certain processing, and then send the control command to the slave station.  If it is read when it is sent, isn't the control instruction we send at this time refer to the slave station state of the previous cycle

ArthurKetels, 2023-03-01:
Perhaps it is time you read about the basic principles of EtherCAT. Slaves do not send packets, they only can add data to a packet that has been transmitted by the master. So the master sends a LRD or LRW packet, when it arrives at the slave it checks if the logical address range matches the slave FMMU configuration. When it does, it inserts the data at the right spot in the packet. This packet then travels through other slaves (who can also insert their piece of data) and lastly it ends up back at the master. The master reads the packet and copies the data from the slaves in the IO buffer. 

Of course the master can decide the timing of the packets so the data will arrive to the application at the right time.

Again, all of this is perfectly explained in the open documents of the ETG.

lhf739181583, 2023-03-02:
Thank you very much for your answer. At present, I have successfully separated LRD and LWR, and achieved satisfactory results. After testing, the above-mentioned lag problem has been avoided. 
The previous problem was that the synchronization points of the synchronization cycle would be confused after separation. 

After trying, it is found that running LRD and LRW each time with a delay of 1 microsecond can solve this problem. 

In my opinion, it takes a while for data to be returned after LWR. If no delay is added, the synchronized ec_DCtime may be when running LRD or LRW, resulting in a chaotic synchronization cycle

lhf739181583, 2023-03-02:
SOEM is a perfect project, and I look forward to separating LRD and LRW in the later stage