OpenEtherCATsociety/SOEM #682: Do you support open mode? I want to connect both master and slave EtherCAT network segments to a standard Ethernet switch

lhf739181583, 2023-02-28:


ArthurKetels, 2023-03-01:
Although EtherCAT supports its packets to be encapsulated in UDP packets, the routing options are very limited because slaves are completely passive. Unless you have a very specific use case I strongly recommend against using Ethernet switches in an EtherCAT topology.

1GuoHaiTong1, 2024-03-27:
hello，my project:  master and slave are connected through the switch, slaveinfo shows that no slaves found. How can I modify it to make it work？
[image]

1GuoHaiTong1, 2024-03-27:
master system：Ubuntu22.04
slave ： UR20-FBC-EC

ArthurKetels, 2024-03-27:
Unless you have special managed switches that can route master traffic to the slave and back based on port connection number this will not work. SOEM uses the MAC numbers in the ethernet header to detect redundancy configurations.  They are not real MAC numbers. And EtherCAT does not use IP numbers, so there is no way for normal automatic routing to work.

Remove the switches.

1GuoHaiTong1, 2024-03-28:
OK, thanks for your reply and explanation

ArthurKetels, 2024-03-28:
If you want an optic fiber connection between master and slave there are media converters for EtherCAT available.
For example Beckhoff CU1521-0010

1GuoHaiTong1, 2024-03-29:
Thanks again for your advice