OpenEtherCATsociety/SOEM #684: Out frame resending in ecx_srconfirm() causes some weird issues

zivke, 2023-03-08:
I'm working on a master application that requires sending a lot of read SDO requests and I'm constantly getting an issue where they start failing after some time (the slave in mailbox is always full).

I have narrowed it down to the [loop](https://github.com/OpenEtherCATsociety/SOEM/blob/master/oshw/linux/nicdrv.c#L579-L595) inside `ecx_srconfirm()` where the "out" frames are always being resent (`ecx_outframe_red()`) if the waiting for the "in" frame (`ecx_waitinframe_red()`) fails in any way.

Now, I'm not 100% sure that this is not an issue on the slave side, but I was wondering if there is any particular reason why this is being done this way - why the "out" frame is being resent in this case?

I have tested the code after moving the `ecx_outframe_red()` function in front of the loop (and therefore sending the "out" frame only once) and everything still seems to be working just fine, but the issue doesn't happen any more.

Any info would be appreciated.
Thanks!

PS Excellent work BTW, this a very useful tool. :muscle:

ArthurKetels, 2023-03-09:
I think you have stumbled on a corner case that is not handled well in SOEM. The packet transmit/receive logic for blocking functions assumes that when a packet is transmitted it should always return back to the master. If there is no packet received it must have been lost somewhere in the stack. The solution is then to retransmit the packet when this is possible within the timeout value. 

Now we have mailbox communication. When a in-mailbox is full it gets processed by the slave, the result is then put in the out-mailbox. If we now fill the in-mailbox again before we empty the out-mailbox the slave behaviour is undefined. a) The slave reads the in-mailbox, processes the data, and waits for the out-mailbox to become empty again. b) The slave waits with reading the in-mailbox until the out-mailbox is empty. It seems you have situation b). 

The simplest solution is to change the timeout value of ecx_FPWR below, from EC_TIMEOUTRET3 in EC_TIMEOUTRET. This will prevent a retransmit if the mailbox send was not successful.

```
int ecx_mbxsend(ecx_contextt *context, uint16 slave,ec_mbxbuft *mbx, int timeout)
{
   uint16 mbxwo,mbxl,configadr;
   int wkc;

   wkc = 0;
   configadr = context->slavelist[slave].configadr;
   mbxl = context->slavelist[slave].mbx_l;
   if ((mbxl > 0) && (mbxl <= EC_MAXMBX))
   {
      if (ecx_mbxempty(context, slave, timeout))
      {
         mbxwo = context->slavelist[slave].mbx_wo;
         /* write slave in mailbox */
         wkc = ecx_FPWR(context->port, configadr, mbxwo, mbxl, mbx, EC_TIMEOUTRET3);
      }
      else
      {
         wkc = 0;
      }
   }

   return wkc;
}
```

zivke, 2023-03-10:
> The simplest solution is to change the timeout value of ecx_FPWR below, from EC_TIMEOUTRET3 in EC_TIMEOUTRET. This will prevent a retransmit if the mailbox send was not successful.

Hm, yes, that solution would also probably work in my case, but essentially it produces the same result as my solution, that the loop inside `ecx_srconfirm()` gets executed only once.

> If there is no packet received it must have been lost somewhere in the stack. The solution is then to retransmit the packet when this is possible within the timeout value.

The issue that I see with this is the one that I think I have experienced:
What if the packet is not lost, just delayed? Or what if the packet made it to the slave, the slave processed it, but then it didn't make it to the master? Then the slave would have the result ready, mailbox full, but the master would send a frame duplicate and potentially mess things up.
[image]

BTW, I'm just brainstorming here in an attempt to maybe further improve the code if anyhow possible.

Thanks for the quick reply.

ArthurKetels, 2023-03-11:
I still think my above patch is fundamentally the best solution.  For almost all cases we want to resend a packet when it did not return to the master (and we have time left). The only exception is ecx_mbxsend(). So that is the location where we should change behavior. 

Now to your wireshark capture. Little nitpick, do not copy a screenshot but attach the cut trace as .pcap file. With the file I can do deep dive in the data when needed. But the marked bit shows exactly what I needed to see. The second marked line shows a delta time over 2ms. This can not be the real propagation delay between tx and rx of the packet, as it is much too long. 

Your real problem is latency in your network stack. From experience this is almost sure to be found in the receive side as the transmit side has inherent low latency (at least for EtherCAT use case).  EtherCAT packet delays are strictly deterministic, and can be measured with high accuracy and confidence. Each slave adds a precise and constant delay between the master tx and rx event. So when a packet is transmitted it is known within a few 10s of ns when the packet is received again. A packet can never arrive late. So when wireshark captures a "late" packet this is due to kernel internal delays after the packet has physically arrived in the NIC. 

There are many methods to be found on the internet how to minimize latency in the network stack.  First, use rt-preempt patched kernel. Second, block the power governor to let the CPU go in deeper sleep modes (wake-up takes a lot of time). Third, optimize NIC driver settings for low latency. Fourth, remove tasks that risk contention on the resources that SOEM uses.

zivke, 2023-03-13:
> The only exception is ecx_mbxsend(). So that is the location where we should change behavior.
When you say we, do you mean that this change should an will be made to the official SOEM code?

Thank you very much for the in-depth analysis and suggestions, it helped a lot.