OpenEtherCATsociety/SOEM #685: EtherCAT Hot Connect

alessiomargan, 2023-03-13:
Hi all,
does SOEM support EtherCAT Hot Connect function as Beckhoff TwinCAT ?
Let me just recap the functionality ... 

* *The principle
The Beckhoff TwinCAT EtherCAT Hot Connect function allows preconfigured sections to be removed from or added to the data traffic before the start or during operation of the system. This can take place by disconnecting/connecting the communication line, switching the device on or off or by some other measure. This is called "flexible topology" or Hot Connect.* *

ArthurKetels, 2023-03-13:
No and yes. Out of the box SOEM does not support hot-connect. Or it does, but half way. 

This is because, at configuration, SOEM reads the slave stack and generates an IO mapping based on the available slaves. Slaves that are attached after this step are simply ignored. But slaves can turn off during operational state and later turn on again. SOEM handles this gracefully. Even a hard reset of a slave is detected and SOEM reprograms the SM and FMMU's on the fly.

What SOEM does not do is map new slaves that where unknown at configuration time. 

If your application knows the different slave configurations beforehand you can fill the slave tree configuration and mapping structure yourself. There is no hard requirement that you have to use all SOEM functions in the same order as in the example applications.  One feature that can really help you here are "groups". Each slave section can be assigned a separate group. And each group has its own IO mapping section that is independent of others. You can even run each group in different tasks at different cycle times.

There are also risks involved. Slave synchronization is very difficult to maintain when slaves appear and disappear at will. Also the recovery and reprogramming of slaves during operational mode costs bandwidth that your application might not have to spare. But can it be made to work reliably? Yes. It has been proven in real world applications with SOEM.

alessiomargan, 2023-03-14:
Hi Arthur,
thanks for the reply.
I'll look at "groups" and try to figure out how to use them.
At the moment we use DC to sync slaves at 1kHz and once in OP a mismatch in the expected WKCs is handled as a bus fault.
Do you know if exist something similar to an application note relative to this topic ?