OpenEtherCATsociety/SOEM #686: call of ec_init() with unknown interface returns "ok"

JMasak, 2023-03-14:
Hi all,
in a small test application i called `ec_init("ETH0")` instead of `ec_init("eth0")` and everything seemed to work fine (return value > 0).

In oshw/linux/nicdrv.c the return values the two calls to ioctl that contain ifname are stored in r but r is not used after storing the result of the call of bind. (Lines 156 - 179)

I guess you could combine all the results (r |= ioctl...) or use an early `return 0`.

ArthurKetels, 2023-03-17:
You are right that there should be a decent fail handling in the ec_init() function. And bailing out early if there is no NIC interface found is certainly one of them. But in testing here with a few computers I could not replicate your behaviour. When ec_init() is called with a unknown name ifr.ifr_ifindex is zero and the bind call will fail. Perhaps your system does not return zero for ifr.ifr_ifindex, or you have a valid interface for zero index. Anyway, a code improvement is needed.

JMasak, 2023-03-17:
If it helps:
I tested this on a Raspberry Pi 3 with the latest release of Raspberry Pi OS (64-Bit) and the "internal" Ethernet interface. wlan0 was connected to my home wlan at the time.