OpenEtherCATsociety/SOEM #687: Lossing slaves in SOEM,but can find using Twincat

SKTTelephoneFaker, 2023-03-20:
I have the same phenomenon,
```
int ecx_recover_slave(ecx_contextt *context, uint16 slave, int timeout)
{
   int rval;
   int wkc;
   uint16 ADPh, configadr, readadr;

   rval = 0;
   configadr = context->slavelist[slave].configadr;
   ADPh = (uint16)(1 - slave);
   /* check if we found another slave than the requested */
   readadr = 0xfffe;
   wkc = ecx_APRD(context->port, ADPh, ECT_REG_STADR, sizeof(readadr), &readadr, timeout);
   if (wkc < 0) {
       printf("wkc < 0 in recover");
   }
```

- It happened when I unplugged the Internet cable
- wkc will always <0
- But when I used twincat, everything worked fine

// Using WIndows10 (64Bit)
// WinPCap 4.1.3
//soem wireshark caps:
[save.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11013146/save.zip)
//twincat wireshark caps:
[twincat_recover.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11013162/twincat_recover.zip)

_Originally posted by @SKTTelephoneFaker in https://github.com/OpenEtherCATsociety/SOEM/issues/99#issuecomment-1475508215_

ArthurKetels, 2023-03-20:
This is not a SOEM issue but something lower in the network stack. Either WinPCap or the windows network stack. As seen in the wireshark trace there is not a single return packet from the slave. If you can verify that the ACT led on your PC or the slave is blinking then you know there is a physical link. One thing that you might try is to use Npcap instead of WinPcap.

SKTTelephoneFaker, 2023-03-21:
Thanks for the quick response, I'll try

SKTTelephoneFaker, 2023-03-22:
[image]
i changed this line and problem seems solved!
@ArthurKetels

closed 2023-03-23 by SKTTelephoneFaker