OpenEtherCATsociety/SOEM #688: General question about ports

bhargavah, 2023-03-20:
Can an external adaptor be used to convert the USB port into an Ethernet port and do EtherCat protocol programming? Since Ethercat protocol works only with the Ethernet port and mac air does not have an Ethernet port, how is it possible to do EtherCat programming on mac air otherwise?

ArthurKetels, 2023-03-20:
Modern USB to Ethernet adapters work fine with SOEM. Just do not expect the same real-time performance as native network cards. The USB part adds quite a bit of timing jitter and latency.

bhargavah, 2023-03-21:
Thanks, Arthur. This answer would help many fellow developers who believe that an Ethernet port is a must which is not the case.

closed 2023-03-21 by bhargavah