OpenEtherCATsociety/SOEM #689: safe_op+error

Edphoenixmax, 2023-03-27:
Hi & Servus,
I tried simple_test under Win10 with other EtherCAT slaver and it worked smoothly, but when I tried the same test under raspberry with weidmuller, the slaver state was always in safe_op +error. But this will not happen when I change other slaver like beckoff. can anyone help me solve this problem ?thanks a lot

Edphoenixmax, 2023-03-27:
I am sorry , I am a student , I am not familiar with this ,I dont know how to create the ESI file of this slave，this is the slaveinfo and the wireshark trace
thank you very much for your help
[simple—test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/11076603/simple.test.txt)
Slaveinfo
Starting slaveinfo
ec_init on ens33 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:UR20-FBC-EC
 Output size: 32bits
 Input size: 120bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000230 ID: 4f911c30 Rev: 00020100
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:   4 F:00010064 Type:3
 SM3 A:2400 L:  15 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   4 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000004 Ll:  15 Lsb:0 Leb:7 Ps:2400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 0e
 CoE details: 2b FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

ArthurKetels, 2023-03-27:
No need to excuse yourself, we all need to learn at some point in time. 

The wireshark trace is not usable in text format. Simply attach the raw .pcap file. Those can be read by the wireshark program. Then I can use the dissector and filters to dive into the slave behaviour.

Edphoenixmax, 2023-03-27:
thanks for your reply, my laptop is in my lab, I will send this as soon as I can 
by the way , may I ask how long have u been studying or working in this field , I am a new student here , I even have to learn C and C ++ first, can you give me some advice or recommand some paper and books about this field.
Soem is more like a flat （I dont know if its approprate to describe this）, I have to make my own C file to with functions in soem to control these slaves . But what can I do with only some IO like EL2088 of BECKOFF

Edphoenixmax, 2023-03-28:
this is the wireshark trace of the simple_test
[simple—test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11084118/simple.test.zip)
thanks a lot

ArthurKetels, 2023-03-28:
Thanks for the .pcap file. This is very helpful. The slave and SOEM happily configure and go to OP mode. Nothing wrong there.

But in your capture it is shown that on regular intervals SOEM is blocked for more than 340ms. Each slave has a default watchdog that times out after approximately 100ms of receiving no master packets. When this triggers the slave goes into safe-OP+error. SOEM detects this and recovers the slave by ack the error and command to OP mode. All is well, until there is another 340ms block, and the problem renews.

So what you have to do is figure out why this block exists on your platform. This forum is not the place to debug OS latency problems. The same for your general questions about programming. I feel your problem, but if I would start with those issues I would need 100 hours per day. SOEM is a library for people that have a good understanding of EtherCAT and programming. All I can do is point you to the samples and documentation that is part of SOEM.

closed 2023-03-28 by ArthurKetels