OpenEtherCATsociety/SOEM #691: SOEM in Visual Studio Code

jonnYyYyYy, 2023-03-31:
Hey Im'm new in coding with VS Code and i have some troubles to implement the SOEM library.

I downloaded the .zip, extracted it and installed CMake.

I did the steps " mkdir build" and "cd build", but at the next step "cmake .. -G "NMake Makefiles"", i get following error:
CMake Warning:
  Ignoring extra path from command line:

   ".."

CMake Error: The source directory "C:/Users/Prüfstand/Documents/Programme" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

ArthurKetels, 2023-04-01:
You need Visual Studio C++, not the code editor Visual Studio Code. 

When you have installed VS C++ then press the "windows" key and look up de Visual Studio folder. There select the command prompt for VS. A command line window opens. Now all your environment variables are set to compile code. 

Then change to the SOEM main folder with cd. And only then follow the steps in the readme.

SKTTelephoneFaker, 2023-04-03:
If you want to set up a development environment, make sure you have the following two options installed.
[image]

jonnYyYyYy, 2023-04-05:
Thanks for your help @ArthurKetels @SKTTelephoneFaker  ! I'm now using Visual Studio.
But when I use the "cmake .. -G "NMake Makefiles"" command I get the followig error message:

PS C:\Users\Prüfstand\Documents\Programme\C_Programm\build> cmake .. -G "NMake Makefiles"
-- The C compiler identification is MSVC 19.35.32216.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx86/x86/cl.exe - broken
CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
  The C compiler

    "C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx86/x86/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/Prüfstand/Documents/Programme/C_programm/build/CMakeFiles/CMakeScratch/TryCompile-8m5w8d

    Run Build Command(s):nmake -f Makefile /nologo cmTC_d7a6a\fast &&   "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_d7a6a.dir\build.make /nologo -L                  CMakeFiles\cmTC_d7a6a.dir\build
    Building C object CMakeFiles/cmTC_d7a6a.dir/testCCompiler.c.obj
        "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E cmake_cl_compile_depends --dep-file=CMakeFiles\cmTC_d7a6a.dir\testCCompiler.c.obj.d --working-dir=C:\Users\Pr³fstand\Documents\Programme\C_programm\build\CMakeFiles\CMakeScratch\TryCompile-8m5w8d --filter-prefix="Hinweis: Einlesen der Datei: " -- C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1435~1.322\bin\Hostx86\x86\cl.exe @C:\Users\PRFSTA~1\AppData\Local\Temp\nm1D6.tmp
    testCCompiler.c
    Linking C executable cmTC_d7a6a.exe
        "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_d7a6a.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100220~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100220~1.0\x86\mt.exe --manifests -- C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1435~1.322\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_d7a6a.dir\objects1 @C:\Users\PRFSTA~1\AppData\Local\Temp\nm225.tmp
    LINK Pass 1: command "C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1435~1.322\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_d7a6a.dir\objects1 /out:cmTC_d7a6a.exe /implib:cmTC_d7a6a.lib /pdb:C:\Users\Pr?fstand\Documents\Programme\C_programm\build\CMakeFiles\CMakeScratch\TryCompile-8m5w8d\cmTC_d7a6a.pdb /version:0.0 /machine:X86 /debug /INCREMENTAL /subsystem:console kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:CMakeFiles\cmTC_d7a6a.dir/intermediate.manifest CMakeFiles\cmTC_d7a6a.dir/manifest.res" failed (exit code 1201) with the following output:
    LINK : fatal error LNK1201: Fehler beim Schreiben in die Programmdatenbank "C:\Users\Pr?fstand\Documents\Programme\C_programm\build\CMakeFiles\CMakeScratch\TryCompile-8m5w8d\cmTC_d7a6a.pdb"". Möglicherweise ist nicht genügend Festplattenspeicher verfügbar, ist der Pfad ungültig oder reichen die Zugriffsrechte nicht aus.
    NMAKE : fatal error U1077: ""C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_d7a6a.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100220~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100220~1.0\x86\mt.exe --manifests -- C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1435~1.322\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_d7a6a.dir\objects1 @C:\Users\PRFSTA~1\AppData\Local\Temp\nm225.tmp": R³ckgabe-Code "0xffffffff"
    Stop.
    NMAKE : fatal error U1077: ""C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_d7a6a.dir\build.make /nologo -L                  CMakeFiles\cmTC_d7a6a.dir\build": R³ckgabe-Code "0x2"
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Prüfstand/Documents/Programme/C_programm/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Prüfstand/Documents/Programme/C_programm/build/CMakeFiles/CMakeError.log".

ArthurKetels, 2023-04-07:
My suggestion is to test with a simple "hello world" program first, to see if you compile chain works. You can google for tutorials that just do that. When that works you can try with SOEM.