OpenEtherCATsociety/SOEM #693: Cannot connect to Beckhoff EP3174

alexanderivanov1112, 2023-04-09:
Hi everyone!

I'm using SOEM with Delta servos, and also with Beckhoff IO modules like EK1100 / EL2008, and everything works fine. 

But I have trouble with working Beckhoff IO module EP3174. Unlike EL2008 (that is very simple), this module has some SDO parameters. 

SOEM is getting stuck while trying to change slave's state to SAFE_OP (ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE * 4) == EC_STATE_SAFE_OP fails on timeout).

Same code with EL2008 works ok.

I've researched on Beckhoff manuals and found out that this module has by default not "DC - Synchron with SYNC0 Event" but "Synchron with SM 3 Event" (sync mode==01). I've tried to write SDO parameter 1C32 (sync mode) after entering in PRE_OP to 01 (sync0) with no success (WKC is 0).

So I don't know how I can correctly work with this module. Is SOEM capable to use SM2 sync? Or I just need do some specific configuration of this module?

I've attached slaveinfo -sdo output as a file.
[analog_input_sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/11186324/analog_input_sdo.txt)

ArthurKetels, 2023-04-10:
I do not think the sync mode is the issue. Sync with SM3 just means the slave tries to line up with the master PDO cycle. As the slave has only inputs (it is a analog input slave) it uses SM3 to detect the master cycle. This means you have to provide a stable master PDO cycle BEFORE you request a change from pre-OP to OP. 

And most likely you just provide a single PDO cycle and then request OP state. The slave now waits for multiple PDO cycles to figure out the synchronization and that never happens because your applications waits for the slave to go in OP, and during that time it is not sending any PDO.

alexanderivanov1112, 2023-04-10:
Thank you very much. That was really cause of problem. I've tried simple_test and it worked fine, after that I found what I did wrong.

closed 2023-04-11 by alexanderivanov1112