OpenEtherCATsociety/SOEM #694: SOEM ec_slave[0].inputs returns zero

wlpeng19911214, 2023-04-12:
the ec_slave[i].PO2SOconfig is this:

int Postionsetup(uint16_t slave)
{
    uint8_t u8val;
    uint16_t u16val;
    uint32_t u32val;

    // assign TxPDO mapping index 0x1a00 to input SM3
    u8val = 0;
    ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u8val = 0;
    ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    //0x1a00
    u16val = 0x1a00;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x60410010;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60640020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x606c0020;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60770010;
    ec_SDOwrite(slave, u16val, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60610010;
    ec_SDOwrite(slave, u16val, 0x05, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60620020;
    ec_SDOwrite(slave, u16val, 0x06, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x606b0020;
    ec_SDOwrite(slave, u16val, 0x07, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60740010;
    ec_SDOwrite(slave, u16val, 0x08, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x603f0010;
    ec_SDOwrite(slave, u16val, 0x09, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x23000010;
    ec_SDOwrite(slave, u16val, 0x0a, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x0a;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    //0x1a01
    u16val = 0x1a01;
    u8val = 0x0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x60410010;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60640020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60610010;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x03;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    //0x1a02
    u16val = 0x1a02;
    u8val = 0x0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x60410010;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x606c0020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60610010;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x03;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    //0x1a03
    u16val = 0x1a03;
    u8val = 0x0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x20040020;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x20050020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x20060020;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x20070020;
    ec_SDOwrite(slave, u16val, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x04;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    //0x1600
    u16val = 0x1600;
    u8val = 0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x60400010;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x607a0020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60ff0020;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60710010;
    ec_SDOwrite(slave, u16val, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60600010;
    ec_SDOwrite(slave, u16val, 0x05, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60b10020;
    ec_SDOwrite(slave, u16val, 0x06, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60b20010;
    ec_SDOwrite(slave, u16val, 0x07, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60720010;
    ec_SDOwrite(slave, u16val, 0x08, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x00000010;
    ec_SDOwrite(slave, u16val, 0x09, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x00000010;
    ec_SDOwrite(slave, u16val, 0x0a, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x0a;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    
    //0x1601
    u16val = 0x1601;
    u8val = 0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x60400010;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x607a0020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60600010;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x03;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    
    //0x1602
    u16val = 0x1602;
    u8val = 0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x60400010;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60ff0020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x60600010;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x03;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    //0x1603
    u16val = 0x1603;
    u8val = 0;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u32val = 0x20000020;
    ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x20010020;
    ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x20020020;
    ec_SDOwrite(slave, u16val, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u32val = 0x20030020;
    ec_SDOwrite(slave, u16val, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
    u8val = 0x04;
    ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    u16val = 0x1600;
    u8val = 1;
    ec_SDOwrite(slave, 0x1c12, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
    u8val = 1;
    ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u16val = 0x1a00;
    ec_SDOwrite(slave, 0x1c13, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
    u8val = 1;
    ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    int16_t i16val = 0x08;
    ec_SDOwrite(slave, 0x6060, 0x00, FALSE, sizeof(i16val), &i16val, EC_TIMEOUTRXM);
    // int16_t tmp2;
    // int size = sizeof(i16val);
    // ec_SDOread(slave, 0x6061, 0x00, FALSE, &size, &tmp2, EC_TIMEOUTRXM);
    // printf("Slave:{%d}, MODE:{%d}\n",slave, tmp2);
    // //8 is for the Postion by 402
    // int16_t i16val = 8;
    // ec_SDOwrite(slave, 0x6060, 0x00, FALSE, sizeof(i16val), &i16val, EC_TIMEOUTRXM);
    // int size = sizeof(i16val);

    // set ESC watchdogdivider
    u16val = 2498;
    ec_FPWR(ec_slave[slave].configadr, 0x0400, sizeof(u16val), &u16val, EC_TIMEOUTRET);
    // set ESC watchdog counter
    u16val = 1000;
    ec_FPWR(ec_slave[slave].configadr, 0x0410, sizeof(u16val), &u16val, EC_TIMEOUTRET);
    // set ESC watchdog counter
    u16val = 1000;
    ec_FPWR(ec_slave[slave].configadr, 0x0420, sizeof(u16val), &u16val, EC_TIMEOUTRET);

    while(EcatError) printf("%s",ec_elist2string());

    return 1;
}

the device can be used if we init it by acontis, step is like this:

1. start the acontis with ENI
2. all of the slaves is in OP state
3. reboot the controller, all the slaves must not be interrupted
4. start the SOEM, and use the config like Postionsetup
5. the wireshark is just like initial_by_actonis_withdcandSDOwrite
[initial_by_actonis_withdcandSDOwrite.txt](https://github.com/OpenEtherCATsociety/SOEM/files/11208088/initial_by_actonis_withdcandSDOwrite.txt)

if the slaves had been interrupted, the wireshark is just like initial_by_soem
[initial_by_soem.txt](https://github.com/OpenEtherCATsociety/SOEM/files/11208093/initial_by_soem.txt)

the ENI of the slave is: 
[eni_txt.txt](https://github.com/OpenEtherCATsociety/SOEM/files/11208113/eni_txt.txt)

ArthurKetels, 2023-04-12:
Please post the raw .pcap files and not some text file. I can only see what is going on when I can use Wireshark to view the captured data.

wlpeng19911214, 2023-04-14:
[pcap_file.zip](https://github.com/OpenEtherCATsociety/SOEM/files/11227901/pcap_file.zip)

sorry, and I post the raw file here.

wlpeng19911214, 2023-04-14:
More infos share:
If I connect only one device with SOEM, the communication is just fine, the pcap file is here.

The error is happened with the first and the second device is ELMO, and the third to the ninth is our own device.